jerryscript-project/jerryscript #4525: Assertion 'ECMA_STRING_IS_REF_EQUALS_TO_ONE (string_p)' in ecma_free_string_list

rerobika, 2021-01-22:
###### JerryScript revision

8ba0d1b6ee5a065a42f3b306771ad8e3c0d819bc

###### Build platform

Ubuntu 18.04.5 LTS (Linux 4.15.0-128-generic x86_64)

###### Build steps

```sh
./tools/build.py --clean --debug --logging=on
```

###### Test case

```js
function* g() {
  class A extends (yield) {}
}
g().next();
```

###### Execution steps & output

```
$ ./build/bin/jerry test.js 
ICE: Assertion 'ECMA_STRING_IS_REF_EQUALS_TO_ONE (string_p)' failed at jerryscript/jerry-core/ecma/base/ecma-literal-storage.c(ecma_free_string_list):77.
Error: ERR_FAILED_INTERNAL_ASSERTION
Aborted (core dumped)
```

ossy-szeged, 2021-10-01:
Note: It is a mem leak, still valid today on 2a82da62add363a33a5df5bc2f2295dd9a440b52

LaszloLango, 2024-12-19:
Let me summarize what we know about this issue:

### Execution Details
When `g().next();` is called:

1. The generator function g begins executing.
2. The `class A extends (yield) {}` line is encountered:
  - The `yield` expression pauses execution and produces a value (which is `undefined` by default when `.next()` is called with no arguments).
  - The generator is now paused, and no further execution happens within `g` until another `.next()` call is made.
3. The generator returns an object `{ value: undefined, done: false }`:
  -  value corresponds to the `yield`'s output (currently undefined because no value is explicitly passed into the `next()` call).
  - `done: false` indicates that the generator is not finished.

At this point:
- **No `class A` is fully defined** because the generator is paused at the yield.
  - The incomplete class definition **remains on JerryScript heap** and won't be freed as it should be.
- To complete the class definition, you’d need to resume the generator with a value for the superclass by calling `.next(SomeClass)`.
  - A complete class definition will be freed from the heap successfully.