jerryscript-project/jerryscript #4532: How to generate bytecode with jerryscript?

QuXing9, 2021-01-23:
Hi! 
As far as I know, when a Js engine executes Javascript code. It will firstly parse the Javascript code into bytecode, then use a intepreter to execute the bytecode. Is this the same for jerryscript? If so, how can I use jerry to get the parsed bytecode?

lygstate, 2021-01-23:
snapshot

rerobika, 2021-01-23:
Hi @QuXing9!

The parsed bytecode is unavaliable from the public API. If you parse a [script](https://jerryscript.net/api-reference/#jerry_parse)/[function ](https://jerryscript.net/api-reference/#jerry_parse_function)you get a runnable/function object which can be [run](https://jerryscript.net/api-reference/#jerry_run)/[invoked ](https://jerryscript.net/api-reference/#jerry_call_function). But if it is not suitable for you could you please describe what would you like to use the native bytecode exactly?

Snapshot is another option. You can pre-parse the script/function and run it later. Please check the [API reference](https://jerryscript.net/api-reference) for more deatiled information.

QuXing9, 2021-01-25:
## Description
Hi, rerobika !

I would like to use the native bytecode to debug the Jerryscript engine.  When I execute the test case in the [attachment](https://github.com/jerryscript-project/jerryscript/files/5864797/testcase.txt), Jerry asserted failure at line 3006 of vm.c. But I don't know the meaning of the opcode `VM_OC_MOV_IDENT`, it looks like judging characters in bytecodes. It is worth noting that Jerry does not report an error when compiling with the parameter `--mem-heap=1572864`. But I don't see an infinite loop or recursion. I am wondering what caused the engine error. 

Looking forward to your reply. Thanks!

 

## JerryScript revision
[Version: 2.4.0 (8be6aec)](https://github.com/jerryscript-project/jerryscript/commit/8be6aec50d599450b82135273e9ca20b0c9ad1ba)

## Build Platform
ubuntu1~16.04.12

## Build Steps:
~~~
python tools/build.py --clean --logging=on --line-info=on --error-messages=on --cpointer-32bit=on --profile=es2015-subset
~~~

## Command
~~~
./jsengines/jerryscript/build/bin/jerry testcase.js
~~~

## Output:
ICE: Assertion 'literal_index < register_end' failed at /export/nisl/qx/jsengines/jerryscript-bd1c4d/jerry-core/vm/vm.c(vm_loop):2369.
Error: ERR_FAILED_INTERNAL_ASSERTION
Aborted (core dumped)

rerobika, 2021-01-25:
You can read about `MOV_IDENT` in #4538. Moveover that's a real issue you found.

rerobika, 2021-01-25:
@QuXing9 I've submitted a PR which resolves your problem.

QuXing9, 2021-01-25:
> @QuXing9 I've submitted a PR which resolves your problem.

Good job!
I am still confused about why the large testcase triggers this error. Could you tell me the reason? Thanks!

rerobika, 2021-01-25:
First of all please check this doc about [bytecode repesentation](https://github.com/jerryscript-project/jerryscript/blob/master/docs/04.INTERNALS.md#compiled-code-format). The large testcase was neccessary since the `NISLFuzzingFunc` has at least 128 bindings (functions/variables) which can be placed into register. This number of bindings reached the limit where a literal index can be encoded in one byte and triggered the failure due to the wrong encoding limit comparison. After the comparison is fixed the literal indices over encoding limit are encoded in two bytes.

PR #4539 (2021-02-04): Fix literal index decoding for CBC_MOV_IDENT
This patch fixes #4532.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
commit a1b682a8 (2021-02-04): Fix literal index decoding for CBC_MOV_IDENT (#4539)

This patch fixes #4532.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
closed 2021-02-04 by dbatyai