jerryscript-project/jerryscript #4538: How to generate CBC_MOV_IDENT

dakhnod, 2021-01-24:
I am trying to manually rebuild some bytecode by trying to reverse engineer the source JS.
While I can find the opcode CBC_MOV_IDENT in the foreign bytecode, I am not able to generate that cbc myself.

Did that code result from a local variable? Or some sort of cache that needs to be turned on on compilation?

I know of this code in v2.1.0
```
if (opcode == CBC_ASSIGN_SET_IDENT && JERRY_LIKELY (first_byte < last_register_index))
        {
          *opcode_pos_p = CBC_MOV_IDENT;
        }
```
but dont't see how that helps me...

zherczeg, 2021-01-24:
Hi,

it is a fast case for a = b, where a is in a register:

function f(b) {
 var a = b;
}

 0 : CBC_EXT_LINE 10
 3 : CBC_PUSH_LITERAL arg:0
 5 : CBC_MOV_IDENT reg:1
 7 : CBC_RETURN_WITH_BLOCK

It is morphed from CBC_ASSIGN_SET_IDENT as you mentioned.

Regards,
Zoltan

2021. január 24. 18:02, "Daniel Dakhno"  írta:
	I am trying to manually rebuild some bytecode by trying to reverse engineer the source JS.
While I can find the opcode CBC_MOV_IDENT in the foreign bytecode, I am not able to generate that cbc myself.

	Did that code result from a local variable? Or some sort of cache that needs to be turned on on compilation?

	I know of this code in v2.1.0

	if (opcode == CBC_ASSIGN_SET_IDENT && JERRY_LIKELY (first_byte < last_register_index)) { *opcode_pos_p = CBC_MOV_IDENT; } 

	but dont't see how that helps me...

dakhnod, 2021-01-25:
Hi, thanks for the quick response. Really helped me.

Could please explain how one can return a literal from the main function?
So that my script returns a literal upon execution?

zherczeg, 2021-01-25:
A literal? A literal is a value in the constant pool, you simply return with it:
```
"abc"
```
Returns the `abc` literal.

dakhnod, 2021-01-25:
thanks, i was talking about a literal since the CBC I am trying to achieve is CBC_RETURN_WITH_LITERAL.
What needs to get returned in fact is a register.
Basically, I want to achieve this result: 
```
CBC_RETURN_WITH_LITERAL  lit 00 (register 0) 
```

dakhnod, 2021-01-25:
to further clarify my problem:
i get the opcode ```CBC_RETURN_WITH_LITERAL  lit 00 (register 0)```
when I define my stuff in a function.
I get a different result when i define everything in the global scope.
Then, the var i create becomes global and receives an identifier instead of going into a register.

What I need is to have my var in a register and return that register in the global scope.

dakhnod, 2021-01-26:
turns out i had to write all of my code as the main script and just generate a function-snapshot, not e regular snapshot.

Thnks for the help everyone.

closed 2021-01-26 by dakhnod