jerryscript-project/jerryscript #4553: Why jerry_get_value_from_error release error/non-error value in different way?

lygstate, 2021-01-29:
https://github.com/jerryscript-project/jerryscript/blob/ba06d492a340bee7d82ff9595851cd278b7bd572/jerry-core/api/jerry.c#L1625-L1642

Seems not consistence

lygstate, 2021-01-29:
Implement
```
inline jerry_value_t AS_JERRY_OBJECT(napi_value nvalue) {
  jerry_value_t val = (jerry_value_t)(uintptr_t) nvalue;
  if (jerry_value_is_error(val)) {
    jerry_value_t err_val = jerry_get_value_from_error(val, false);
    jerry_release_value(err_val);
    return err_val;
  }
  return val;
}
```

rerobika, 2021-02-04:
It's simply about internal reference count maganement. The if the `value` is not error, and should be released we just don't put reference for it. If the value should not be released, it means it is needed later, so it kept alive with `ecma_copy_value `. 

> For implement napi_value to object

In the referenced code

```c
jerry_value_t err_val = jerry_get_value_from_error(val, false);
jerry_release_value(err_val);
```

is equivalent to

galpeter, 2021-02-04:
Side note on the following code:
```
    jerry_value_t err_val = jerry_get_value_from_error(val, false);
    jerry_release_value(err_val);
    return err_val;
```
the release_value here will release the error object which you are returning. This seems incorrect as the err_val at the time of return would not have a reference (it was released)

lygstate, 2021-02-04:
> Side note on the following code:
>
> ```
>     jerry_value_t err_val = jerry_get_value_from_error(val, false);
>     jerry_release_value(err_val);
>     return err_val;
> ```
>
> the release_value here will [...]

Yeap, I am confusing about the error object, and didn't got the clue how to using it correctly without memory leak.