jerryscript-project/jerryscript #4554: Uncaught RangeError causes stack overflow

paintedveil5, 2021-01-30:
###### JerryScript revision
e191a5cf47eeaf3e36098e49a138885f913c0dd1
(latest master - 2021.01.29.)

###### Build platform
Ubuntu 16.04.1 (Linux 4.15.0-117-generic x86_64)

###### Build steps
```sh
python ./tools/build.py --clean --debug --compile-flag=-fsanitize=address --compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer  --compile-flag=-fno-common  --lto=off --error-message=on --system-allocator=on --compile-flag=-Wno-uninitialized --compile-flag=-Wno-missing-field-initializers --link-lib="m"
```
###### Test case
```
// test.js
a= function() { b = function() { a();}();}
b = function() { a(); }();
```
###### Output
```
ASAN:DEADLYSIGNAL
=================================================================
==3701==ERROR: AddressSanitizer: stack-overflow on address 0xff08aef4 (pc 0x0813006d bp 0xff08b318 sp 0xff08aee0 T0)
    #0 0x813006c in malloc (/home/lily/Desktop/crash-analyze/jerry-crash+0x813006c)
    #1 0x828c19e in jmem_heap_alloc /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/jmem/jmem-heap.c:254:10
    #2 0x828c19e in jmem_heap_gc_and_alloc_block /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/jmem/jmem-heap.c:298
    #3 0x81c5133 in ecma_create_object /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/ecma/base/ecma-helpers.c:79:38
    #4 0x822bc86 in ecma_op_create_function_object /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:366:27
    #5 0x822bee2 in ecma_op_create_any_function_object /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:590:10
    #6 0x83198e9 in vm_construct_literal_object /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/vm/vm.c:486:18
    #7 0x83075a6 in vm_loop /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/vm/vm.c:1085:9
    #8 0x82f1baf in vm_execute /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/vm/vm.c:4948:37
    #9 0x82f1204 in vm_run /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/vm/vm.c:5055:10
    #10 0x822dfa7 in ecma_op_function_call_simple /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1147:15
    #11 0x822d7ab in ecma_op_function_call /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1377:14
    #12 0x82f2372 in opfunc_call /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/vm/vm.c:792:24
    #13 0x82f2372 in vm_execute /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/vm/vm.c:4954
    #14 0x82f1204 in vm_run /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/vm/vm.c:5055:10
    #15 0x822dfa7 in ecma_op_function_call_simple /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1147:15
    #16 0x822d7ab in ecma_op_function_call /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1377:14
    #17 0x82f2372 in opfunc_call /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/vm/vm.c:792:24
    #18 0x82f2372 in vm_execute /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/vm/vm.c:4954
    #19 0x82f1204 in vm_run /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/vm/vm.c:5055:10
    #20 0x822dfa7 in ecma_op_function_call_simple /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1147:15
    #21 0x822d7ab in ecma_op_function_call /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1377:14
    #22 0x82f2372 in opfunc_call /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/vm/vm.c:792:24
    #23 0x82f2372 in vm_execute /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/vm/vm.c:4954
    #24 0x82f1204 in vm_run /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/vm/vm.c:5055:10
    #25 0x822dfa7 in ecma_op_function_call_simple /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1147:15
    #26 0x822d7ab in ecma_op_function_call /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1377:14
    #27 0x82f2372 in opfunc_call /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/vm/vm.c:792:24
    #28 0x82f2372 in vm_execute /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/vm/vm.c:4954
    #29 0x82f1204 in vm_run /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/vm/vm.c:5055:10
    #30 0x822dfa7 in ecma_op_function_call_simple /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1147:15
    #31 0x822d7ab in ecma_op_function_call /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1377:14
    #32 0x82f2372 in opfunc_call /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/vm/vm.c:792:24
    #33 0x82f2372 in vm_execute /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/vm/vm.c:4954
    #34 0x82f1204 in vm_run /home/ubuntu/lastest-jerryscript/jerryscript/jerry-core/vm/vm.c:5055:10
...
```
###### Expected behavior
Please handle the situation that circular references exist in a javascript file and throw a RangeError.

akosthekiss, 2021-01-30:
Please, try to build the engine with stack limit enabled (--stack-limit=NNN). A stack overflow/infinite recursion is only an issue if reproducible with a set stack limit.

paintedveil5, 2021-01-30:
> Please, try to build the engine with stack limit enabled (--stack-limit=NNN). A stack overflow/infinite recursion is only an issue if reproducible with a set stack limit.

For --stack-limit=NNN, how much NNN should be set?

akosthekiss, 2021-01-30:
E.g., you can try 16.

rerobika, 2021-02-01:
The problem is not present with `--stack-limit=16` so closing the issue.

closed 2021-02-01 by rerobika