jerryscript-project/jerryscript #4563: Question: what does ERR_DISABLED_BYTE_CODE mean?

haochenli, 2021-02-04:
I am so sorry because I CAN NOT post the code in public, which will make it everybody difficult to get my point.
Basically what I did is:

1. using `jerry_generate_function_snapshot` to create a snapshot.
2. using `jerry_load_function_snapshot` to load the snapshot.
3. using `jerry_call_function` to execute the function created in step2.

Everything works well in the [gtest](https://github.com/google/googletest) framework.
But when I switch into [benchmark](https://github.com/google/benchmark) framework with the same code. The error `ERR_DISABLED_BYTE_CODE` popup. I try to google it but get zero message about this error.
Could you please give me a hint of what will cause the error `ERR_DISABLED_BYTE_CODE`? And what does this error mean?

The template in the benchmark is like 
```c
 static void initInfras(benchmark::State& state) {
 // more like a sudo code below...
    jerry_generate_function_snapshot();
    jerry_load_function_snapshot();
    for (auto _ : state) {
        jerry_call_function();
    }
BENCHMARK(initInfras);
```

rerobika, 2021-02-04:
The most common occurrence of `ERR_DISABLED_BYTE_CODE ` is when the engine configuration does not match properly during the snapshot generation and the execution. E.g.: you generate a snapshot when the line info support is active, but you run this code with an other build when this feature is disabled. In this case all the line info releated VM instructions are goind to raise disabled bytecode error.

Q: Could you ensure that neither `jerry_generate_function_snapshot` or `jerry_load_function_snapshot` returns with error?

haochenli, 2021-02-05:
Thanks for your quick response sir!
I am **SURE** that `jerry_generate_function_snapshot ` and `jerry_load_function_snapshot ` **NOT** return errors because I check the return value in my code with `jerry_value_is_error` function after these executions.
And thank you again, I m gonna check the configuration in my project.
I will post the follow-up info if I have any progress.

haochenli, 2021-02-05:
By the way, I like your Avatar picture. that's so funny :) @rerobika

zherczeg, 2021-02-05:
Then the problem is exactly what was described by @rerobika. The engine cannot check the vailidity of the byte code, it trust you, so it returns with no error during loading. But it turns out in this case, that some instructions cannot be executed, so you used different compile options for compiling the snapshot generator and the loader program.

haochenli, 2021-02-05:
Thank you guys @zherczeg @rerobika.
Update:  As they said, the error will occur if you use different compile options for `creating and loading snapshot` and `call the snapshot`, which caused by [benchmark](https://github.com/google/benchmark).
The code change from:
```c
static void initInfras(benchmark::State& state) {
 // more like a sudo code below...
    jerry_generate_function_snapshot();
    jerry_load_function_snapshot();
    for (auto _ : state) {
       // I try to put log here and figure out that some time the `jerry_call` will happened without previously generate and load.
        jerry_call_function();
    }
BENCHMARK(initInfras);
```
to
```c
 // more like a sudo code below...
    for (auto _ : state) {
        jerry_generate_function_snapshot();
        jerry_load_function_snapshot();
        jerry_call_function();
    }
BENCHMARK(initInfras);
```
Then the error disappeared. Seems like in the `for` loop, there is a 'dismatch' happened.
Actually, I am a javascript guy, so I am not gonna dig into the reason why this happens. 
Thank you guys again.

rerobika, 2021-02-05:
You're welcome.

closed 2021-02-05 by rerobika