jerryscript-project/jerryscript #4590: Snapshot error: RangeError: Unsupported static snapshot literal: 2147483647

lygstate, 2021-02-10:
###### JerryScript revision
34a6a93597e5b84ad0e892039df31885c76fc4e4

###### Build platform
Name the build platform. E.g., copy the output of
`echo "$(lsb_release -ds) ($(uname -mrs))"` (on Linux),
`echo "$(sw_vers -productName) $(sw_vers -productVersion) ($(uname -mrs))"` (on macOS), or
`python -c "import platform; print(platform.platform())"` (should work everywhere).

###### Build steps
Describe how to build JerryScript. Give all the necessary details of the build
(e.g., environment variables, command(s), profile, command line options, etc.).

E.g.:
```sh
python tools/build.py --snapshot-exec=ON --snapshot-save=ON --jerry-cmdline-snapshot=ON

```

###### Test case
creating static snapshot for test.js
```
function a() {
	return 0x7FFFFFFF
}

function b() {
	return 0xFFFFFFFF
}

function c() {
	return 3.4028234663852886e+38
}
```
Test step:
```
export PATH=$PATH:$PWD/build/bin/
jerry-snapshot generate  -o test.snapshot test.js
jerry-snapshot litdump -o test.list test.snapshot
jerry-snapshot generate  -o test.final.snapshot --static --load-literals-list-format test.list test.js

```

###### Output
```
lygstate@DESKTOP-94PU0GB:/mnt/c/work/study/languages/typescript/jerryscript$ jerry-snapshot generate  -o test.final.snapshot --static --load-literals-list-format test.list test.js
Input file 'test.js' (133 bytes) loaded.
Input file 'test.list' (22 bytes) loaded.
Error: Generating snapshot failed!
Snapshot error: RangeError: Unsupported static snapshot literal: 2147483647
```

LaszloLango, 2021-03-23:
This is an unfortunate limitation of static snapshots. You need to use a trick for your (big) numbers. Try to do it like this:

```js
function a() {
	return '0x7FFFFFFF' - 0
}

function b() {
	return '0xFFFFFFFF' - 0
}

function c() {
	return '3.4028234663852886e+38'  - 0
}
```

lygstate, 2021-03-24:
> This is an unfortunate limitation of static snapshots. You need to use a trick for your (big) numbers. Try to do it like this:
> 
> ```js
> function a() {
> 	return '0x7FFFFFFF' - 0
> }
> 
> function b() {
> 	return '0xFFFFFFFF' - 0
> }
> 
> function c() {
> 	return '3.4028234663852886e+38'  - 0
> }
> ```

Well, that's weird. and bloks me using jerryscript

LaszloLango, 2021-03-24:
> Well, that's weird. and blocks me to using jerryscript
I am sorry to hear that. Maybe @zherczeg can help you to fix this inside the engine if my suggested workaround does not fit into your use case.

zherczeg, 2021-03-24:
Static snapshots has many limitations. They need to run without using jerry heap, so only direct values can be used. The suggestions above requres more external magic strings, but you often don't need them. E.g.
`0xFFFFFFFF = -1 >>> 0`
`0x7FFFFFFF = -1 >>> 1`

lygstate, 2021-03-24:
> Static snapshots has many limitations. They need to run without using jerry heap, so only direct values can be used. The suggestions above requres more external magic strings, but you often don't need them. E.g.
> `0xFFFFFFFF = -1 >>> 0`
> `0x7FFFFFFF = -1 >>> 1`

Does this are the limitation of `jerryscript compact bytecode`, any plan to fix this limitation?

zherczeg, 2021-03-24:
Static snapshots are executed directly from read-only memory, so their literal pool cannot be modified. The literal pool cannot have pointers, because they would point to wrong values after loading. If you need generic values, use normal snapshots, which loads the literal pool into the memory and sets the pointers corectly (the byte code can remain in read-only memory). Static snapshots are designed for extremely constrained devices.

lygstate, 2021-03-24:
> Static snapshots are executed directly from read-only memory, so their literal pool cannot be modified. The literal pool cannot have pointers, because they would point to wrong values after loading. [...]

I know of that, I've already read the full document of jerryscript design about `compact byte code`.
I think we can encode these literals(double) as offset(not pointer) with simple value.

zherczeg, 2021-03-24:
But how does vm knows that it is an offset when the vm executes the code? Creating new values every time when a literal is loaded can be costly as well.

lygstate, 2021-03-24:
> But how does vm knows that it is an offset when the vm executes the code? Creating new values every time when a literal is loaded can be costly as well.

the offset is encoded in jerry_value_t, of-cause it's knows. unless there is toooo much double literals
```
The major structure for data representation is ECMA_value. The lower three bits of this structure encode value tag, which determines the type of the value:

simple
number
string
object
symbol
error
```
The simple occupy 29bit, and only have
```
undefined
null
true
false
empty (uninitialized value)
```
we have 2^29-3 values to encode with `simple` value, we can create a double literal store with `simple` value.

rerobika, 2021-12-06:
Conclusion: That's the static snapshot's limitation, not a bug.

closed 2021-12-06 by rerobika

lygstate, 2021-12-06:
> Conclusion: That's the static snapshot's limitation, not a bug.

Can we fixes this limitation?

zherczeg, 2021-12-07:
The plan is dropping static snapshot support. That solves this.