jerryscript-project/jerryscript #4608: Questions: How does snapshot speed up the process?

haochenli, 2021-02-22:
I read the article about  [V8's snapshot](https://v8.dev/blog/custom-startup-snapshots). AFAIK the snapshot in v8 means initialized context with prepared heap snapshot which was created at the very beginning, that will speed up the bootstrap process.

And in Jerry's doc and test file(test-snapshot.c), what I got is the snapshot is more like a "faster code" which is actually bytecode.(question1: Am I right about the understanding of the snapshot?)

So I write a simple Fibonacci test to check it out:
Here is js: 
```js
function fibonacci(num) {
    if (num <= 1) return 1;
  
    return fibonacci(num - 1) + fibonacci(num - 2);
}

print(fibonacci(40))
```

And then I generate the snapshot with the cmd tool: `/jerryscript/build/bin/jerry-snapshot' generate '/Users/lihaochen/Desktop/Project/os/js-engine-all/test.js' -o js1.snapshot`.

And the next step I try to compare the execution time between with snapshot and without snapshot. 
Here is the result:
[image]
Finally, the execution time is almost the same...(question2: where am I wrong?)

I have already added  `JERRY_SNAPSHOT_EXEC` and `JERRY_SNAPSHOT_SAVE` config
I appreciate your answer...

zherczeg, 2021-02-23:
Maybe the wording is wrong. The snapshot skips parsing (time and memory saving), but the resulting byte code is the same. No extra optimizations are done on snapshots.

haochenli, 2021-02-23:
> Maybe the wording is wrong. The snapshot skips parsing (time and memory saving), but the resulting byte code is the same. No extra optimizations are done on snapshots.

Thanks for your reply, what is `wording` mean?

zherczeg, 2021-02-23:
https://www.merriam-webster.com/dictionary/wording

haochenli, 2021-02-23:
And I trya benchmark.js file which is the implementation of the Richards benchmark. The result of comparison between with snapshot and without snapshot as below: still the same..
[image]
[Code here](https://github.com/haochenli/Learning/blob/master/benchmark.js)