jerryscript-project/jerryscript #4617: A possible performance problem about String.prototype.toUpperCase

YaoHouyou, 2021-03-04:
#### Description 

When a long string is repeatedly converted to uppercase, the execution time of the Jerryscript much longer than other engines. For example,  [XS](https://github.com/Moddable-OpenSource/moddable) , similar to jerryscript,  a JavaScript engine that runs on resource-constrained devices. Does this problem need to be optimized?

#### Version
bd1c4df

#### Testcase

```javascript
var NISLFuzzingFunc = function (x) {
    x.toUpperCase();
};
var NISLParameter0 = 'PPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPPP';
for (var INDEX = 0; INDEX < 10000; INDEX++) {
    NISLFuzzingFunc(NISLParameter0);
}
```
#### Command
``` bash
build/bin/jerry testcase.js
```

#### Execution time

JerryScript: 5834ms 
[XS](https://github.com/Moddable-OpenSource/moddable)：81ms
[duktape](https://github.com/svaarala/duktape): 103ms 
[quickJS](https://bellard.org/quickjs/):  94ms
MuJS:  320ms
hermes:  47ms

### Build command
``` bash
python2 tools/build.py --clean --logging=on --line-info=on --error-messages=on --cpointer-32bit=on --mem-heap=1572864 --profile=es2015-subset
```
#### CPU

```
Intel(R) Core(TM) i9-9940X CPU @ 3.30GHz
```

rerobika, 2021-03-12:
Closing due to the same reason i #4621 .

closed 2021-03-12 by rerobika

ossy-szeged, 2021-03-12:
Just a note, the runtime was 5800ms on the mentioned year old hash, now it is 130 ms on the latest master, there is no performance issue.

YaoHouyou, 2021-03-12:
@ossy-szeged Thank you.