jerryscript-project/jerryscript #4619: I could not find the definition of the jerry_call_info_t type!

mhungssy, 2021-03-05:
Hello,

Im try to understand jerryscript by build and run example from: https://github.com/jerryscript-project/jerryscript/blob/master/docs/03.API-EXAMPLE.md  And I face 2 problems:

First is: I can not find libjerry-math after build
Second: From example 4, gcc show error: unknown type name ‘jerry_call_info_t’

Please help me build this, I am very interested in this project

galpeter, 2021-03-05:
Hi,

It seems there is a bit of incorrect information on the examples page: that is the jerry-math lib is not built by default. So you can leave it out and instead use the system's libmath (-lm) for now. Thus the build command changes to:
```sh
$ gcc api-example-4.c -o api-example-4 -lm $(pkg-config --cflags --libs libjerry-core libjerry-port-default)
```

Regarding the `jerry_call_info_t` could you share your full error output, and the command which you used to compile it? I've just did a fresh checkout and tried out the build commands (aside from the above mentioned math lib problem) and it compiled ok. In addition could you please specify which git hash are you using to build the examples?

mhungssy, 2021-03-05:
Thank for you quick reply!

Im got follow error:

[image]

galpeter, 2021-03-05:
Ahh I see the problem, you are using JerryScript 2.4.0 however the api docs on the official site is the up-to date (dev) version which is now prior 3.0. The call argument will/was be changed. Please check the api example md file in your source to see the correct arguments for jerry 2.4.0.

closed 2021-03-05 by mhungssy

zherczeg, 2021-03-05:
Note: we are int he middle of redesigning the api so expect a lot of large changes.