jerryscript-project/jerryscript #4621: A problem about String.prototype.search

YaoHouyou, 2021-03-09:
#### Version:

bd1c4df

#### Overview:

​    When executing the following test case, all engines except jerryscript output `-1`. What is the reason for this? In addition, I would like to know what the argument `this` in the jerryscript exception information refers to. If it is `String.prototype`, no exception should be thrown regardless of the definition of ES5.1 or ES6. If not, what does this refer to? 

#### Testcase:

``` javascript
var foo = function(){
	var a = String.prototype.search(1);
	print(a);
};
foo();
```

#### Actual Results:

```
Script Error: TypeError: Argument 'this' is not a String or a String object.
```

#### Expected Results:

```
-1
```

rerobika, 2021-03-12:
The related commit is almost one year old. You have been asked several times before to not report outdated issues. Please either update the revision you are testing, or validate it on the current master before reporting, otherwise we must consider these reports as spam and will act accordingly.

closed 2021-03-12 by rerobika

ossy-szeged, 2021-03-12:
just a note: I got -1 on the mentioned one year old hash and on the latest master too.

YaoHouyou, 2021-03-12:
@ossy-szeged 
For the version bd1c4df, buid jerry with the following command can reproduce it.
```
python2 tools/build.py --clean --logging=on --line-info=on --error-messages=on --cpointer-32bit=on --mem-heap=1572864 --profile=es2015-subset
```
However, when I build jerry with `python2 tools/build.py --clean`, it cannot be reproduced.

ossy-szeged, 2021-03-12:
@YaoHouyou: I bisected for you, 281635cb40648103b538e33611e2a158e598def3 fixed this issue long long time ago. It wasn't a bug, but incompatible changes in ECMAScript specifications. Previously the implementation was conformed to ES2015 spec, but since that change, it conforms to the latest spec.