jerryscript-project/jerryscript #4624: jerryscript can not parse new line \r\n in Windows

mhungssy, 2021-03-11:
Hi,

Almost Windows text editor used "\r\n" to determine new line. When Im trying to load javascript from file in Windows, function jerry_parse return error with js file (using \r\n), when Im replace "\r\n" by "\n" it ok.

My question is can we parse "\r\n" with special flag or I must replace "\r\n" before call parse

Thanks,

zherczeg, 2021-03-11:
\r\n is defined as newline and it is parsed here:

https://github.com/jerryscript-project/jerryscript/blob/master/jerry-core/parser/js/js-lexer.c#L252

Is the following define correct in your case?
`#define LIT_CHAR_CR ((ecma_char_t) 0x000D) /* carriage return */`

mhungssy, 2021-03-11:
That my fault

Thank you,

closed 2021-03-11 by mhungssy