jerryscript-project/jerryscript #4638: draft a new release

gengjiawen, 2021-03-29:
We need some fixes after 2.4.0. Is it okay for your guys to draft a new release ? Thx.

zherczeg, 2021-03-29:
We don't usually do bugfix releases, mostly because of lack of resources. What do you need exactly?

akosthekiss, 2021-03-29:
Let me add some extra detail. The project follows semver, that's the main reason why bugfix/patch releases are uncommon. Usually, when a new release comes out, it already contains new features in addition to bugfixes, so we bump the minor version instead of the patch level, as per semver. What we don't do is backporting and maintenance of old releases (i.e., all development happens on master, there are no release branches) -- that would not be worth the costs.

@gengjiawen BTW, who is 'we'?

gengjiawen, 2021-03-30:
> What do you need exactly?

Mainly https://github.com/jerryscript-project/jerryscript/pull/4529 and https://github.com/jerryscript-project/jerryscript/pull/4539

> BTW, who is 'we'?

Compiler learner :)

Any chance we make release more agile. V8 has a release cycle in 4 weeks.

closed 2021-06-01 by gengjiawen