jerryscript-project/jerryscript #4671: Operator Overloading as proposed or like QuickJS

kungfooman, 2021-05-14:
Hello, I am currently looking for an embeddable JS engine and one concern is Operator Overloading: https://github.com/tc39/proposal-operator-overloading/

There is one engine that supports it: https://bellard.org/quickjs/jsbignum.html#Operator-overloading

Which is highly useful for everything related to math (and microcontrollers/robots control their environment via math).

Is this a planned feature or is there some support for it already?

ossy-szeged, 2021-06-06:
JerryScript implements only features are in the latest ECMAScript specification, sometimes an important and stable stage 3 or stage 4 proposals too. As I see, operator overloading is in a very early stage, it is a stage 1 proposal. I think it's too early to discuss about supporting this feature, let's get back once it become a stage 3 proposal in the future.

rerobika, 2021-12-08:
Please reopen if the proposal gets to stage 3 or higher.

closed 2021-12-08 by rerobika