jerryscript-project/jerryscript #4683: A question about RegExp timeout

YuanWangC, 2021-06-04:
When the regular expression matching operation is performed using regular expression parameter `/([\w]+)*#/` and string parameter `aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa`, the execution time exceeds 40 minutes when the following test case is executed. And when I use PHP language to match with the same parameters, I do not find any timeout. Is this an issue of jerry? 

#### version

```
e8ed543
```

#### command

```
jerryscript/build/bin/jerry testcase.js
```

#### testcase

```javascript
var matchTest = function(regex, testString){
     result = String.prototype.match.call(testString, regex);
         if (result !== null){
                return result[0];
         }
        else return "";
};
var regex = /([\w]+)*#/;
var testString ="aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa";
var result = matchTest(regex, testString);
print(result);
```

zherczeg, 2021-06-04:
Your regexp is badly written, `/\w+#/` matches the same thing with far less execution time. It is the same thing that comparing bubble sort in C and quicksort in python, and complaining that C is slow.

Regarding php, it uses the PCRE2 engine, and it has an optimization which checks `#` early, and since the input has no `#`, it fails early. Try `/([\w]+)*##/` on `"aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa#"` and you will get a slowdown as well (it will be still faster due to JIT compiler optimization).

YuanWangC, 2021-06-08:
I tested the testcase below on the V8 engine and the execution time was less than 3 minutes. Compared to that, jerry execution time is too long. Will this behavior cause some performance problems for jerry?
```javascript
var matchTest = function(regex, testString){
     result = String.prototype.match.call(testString, regex);
         if (result !== null){
                return result[0];
         }
        else return "";
};
var regex = /([\w]+)*#/;
var testString ="aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa";
var result = matchTest(regex, testString);
print(result);
```
And I used the recommended regular expression `/([\w]+)*##/` and string `aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa#` for testing, but it still ran for 45 minutes.
```javascript
var matchTest = function(regex, testString){
     result = String.prototype.match.call(testString, regex);
         if (result !== null){
                return result[0];
         }
        else return "";
};
var regex = /([\w]+)*##/;
var testString ="`aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa#";
var result = matchTest(regex, testString);
print(result);
```

rerobika, 2021-12-08:
I think the comparison to v8 makes no sense. The behavior is correct, since it's a really time consuming regexp for the engine, it takes as long it takes.

closed 2021-12-08 by rerobika