jerryscript-project/jerryscript #4729: ICE: Assertion 'ECMA_STRING_IS_REF_EQUALS_TO_ONE (string_p)' failed  error

X-Crack, 2021-08-05:
`
                            var array = new Array();

                            array[0] = "aaa";
                            array[1] = "aaa";
`

ICE: Assertion 'ECMA_STRING_IS_REF_EQUALS_TO_ONE (string_p)' failed at D:\developdirectory\3rdparty\libjerryscript\jerry-core\ecma\base\ecma-literal-storage.c(ecma_free_string_list):77.
Error: ERR_FAILED_INTERNAL_ASSERTION

galpeter, 2021-08-05:
Hi!

Could you provide some information on this? Like:
* What is the the git hash of jerryscript?
* Is this on Windows? (based on the path)
* How was jerryscript configured? Which features were enabled/disabled?

For me on Linux with the current master (b6ec9275aa44a3c332cd2f37bb87a6677935456f) this is not reproducable.

X-Crack, 2021-08-06:
> 你好！
> 
> 你能提供一些这方面的信息吗？像：
> 
> * jerryscript 的 git 哈希值是多少？
> * 这是在 Windows 上吗？（基于路径）
> * jerryscript 是如何配置的？启用/禁用了哪些功能？
> 
> 对于我在 Linux 上使用当前主 ( [b6ec927](https://github.com/jerryscript-project/jerryscript/commit/b6ec9275aa44a3c332cd2f37bb87a6677935456f) ) 来说，这是不可复制的。

https://github.com/jerryscript-project/jerryscript/releases/tag/v2.4.0

galpeter, 2021-08-06:
I've tried with v2.4.0 on Linux with default configuration, there was no error for this few lines of JS code. My other questions still stand:
* How was jerryscript configured? Which options/features were enabled/disabled?
* For your jerryscript build running this script with the jerry binary reports the same error?
 
More details would be really helpful.

X-Crack, 2021-08-06:
> I've tried with v2.4.0 on Linux with default configuration, there was no error for this few lines of JS code. My other questions still stand:
>
> * How was jerryscript configured? Which [...]

```
#include <string>
#include "jerryscript.h"
int main()
{

    jerry_init(JERRY_INIT_EMPTY);

    std::string src = R"(
        var array = new Array();
        array[0] = "aaa";
        array[1] = "aaa";
    )";

    jerry_value_t f = jerry_eval((const jerry_char_t *)src.data(), src.size(), 0);
    jerry_cleanup();

    return 0;
}
```

cmake all default settings

Windows This code will report an error.

And array array will be ignored if it encounters string fields.

E.g

var array = new Array()
array["aaa"] = "hello";
array["bbb"] = "world";

String fields cannot be traversed.

galpeter, 2021-08-09:
For the example code:
```c
jerry_value_t f = jerry_eval((const jerry_char_t *)src.data(), src.size(), 0);
jerry_cleanup();
```

The `f` value is not released. API users must release `jerry_value_t` after they are no longer needed (see the api ref. documentation). The `jerry_cleanup` will not remove leftover references, but report an error.

Try adding a `jerry_release_value(f);` just before the cleanup call. I've tested this by adding this call and it fixes the error.

Regarding the "array array will be ignored":

How are you trying to traverse the array?
I've tried the following JS script with the jerry binary.
```js
var array = new Array();
array[0] = "aaa";
array[1] = "dd";
array['aa'] = "cc";

for (var idx in array) {
    print(idx + " -> " + array[idx]);
}
```
And the result is the following output on the terminal:
```
0 -> aaa
1 -> dd
aa -> cc
```

X-Crack, 2021-08-10:
> For the example code:
>
> ```c
> jerry_value_t f = jerry_eval((const jerry_char_t *)src.data(), src.size(), 0);
> jerry_cleanup();
> ```
>
> The `f` value is not released. API users must release `jerry_value_t` [...]

ok tks.

X-Crack, 2021-08-10:
jerry_value_t
jerry_json_stringify (const jerry_value_t input_value)

Function provides only one argument.

JavaScript stringify has 3 parameters by default.
JSON.stringify(value, replacer , space) 
**The behaviour is equivalent with the "JSON.stringify(input_value)" JS call.**

galpeter, 2021-08-10:
I think this json stringify comment is not related to this PR request. In addition I don't follow what you what with this comment. Please open another issue or if this is an API related change request, maybe add a comment in the api update proposal issue.

closed 2021-08-11 by rerobika