jerryscript-project/jerryscript #4750: error LNK2019: unresolved external symbol

autoantwort, 2021-08-25:
###### JerryScript revision
git master

###### Build platform
windows 

###### Build steps
vcpkg install

###### Build log
```
[216/221] cmd.exe /C "cd . && D:\downloads\tools\cmake-3.21.1-windows\cmake-3.21.1-windows-i386\bin\cmake.exe -E vs_link_dll --intdir=jerry-core\CMakeFiles\jerry-core.dir --rc=C:\PROGRA~2\WI3CF2~1\10\bin\100190~1.0\x64\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100190~1.0\x64\mt.exe --manifests  -- C:\PROGRA~2\MICROS~2\2019\ENTERP~1\VC\Tools\MSVC\1429~1.301\bin\Hostx64\x64\link.exe  @CMakeFiles\jerry-core.rsp  /out:bin\jerry-core.dll /implib:lib\jerry-core.lib /pdb:bin\jerry-core.pdb /dll /version:0.0 /machine:x64 /nologo    /debug /INCREMENTAL  && cd ."
FAILED: bin/jerry-core.dll lib/jerry-core.lib 
cmd.exe /C "cd . && D:\downloads\tools\cmake-3.21.1-windows\cmake-3.21.1-windows-i386\bin\cmake.exe -E vs_link_dll --intdir=jerry-core\CMakeFiles\jerry-core.dir --rc=C:\PROG
RA~2\WI3CF2~1\10\bin\100190~1.0\x64\rc.exe --mt=C:\PROGRA~2\WI3CF2~1\10\bin\100190~1.0\x64\mt.exe --manifests  -- C:\PROGRA~2\MICROS~2\2019\ENTERP~1\VC\Tools\MSVC\1429~1.301\bin\Hostx64\x64\link.exe  @CMakeFiles\jerry-core.rsp  /out:bin\jerry-core.dll /implib:lib\jerry-core.lib /pdb:bin\jerry-core.pdb /dll /version:0.0 /machine:x64 /nologo    /debug /INCREMENTAL  && cd ."
LINK Pass 1: command "C:\PROGRA~2\MICROS~2\2019\ENTERP~1\VC\Tools\MSVC\1429~1.301\bin\Hostx64\x64\link.exe @CMakeFiles\jerry-core.rsp /out:bin\jerry-core.dll /implib:lib\jerry-core.lib /pdb:bin\jerry-core.pdb /dll /version:0.0 /machine:x64 /nologo /debug /INCREMENTAL /MANIFEST /MANIFESTFILE:jerry-core\CMakeFiles\jerry-core.dir/intermediate.manifest jerry-core\CMakeFiles\jerry-core.dir/manifest.res" failed (exit code 1120) with the following output:
jerry.c.obj : error LNK2019: unresolved external symbol jerry_port_module_resolve referenced in function jerry_module_link
ecma-builtin-date.c.obj : error LNK2019: unresolved external symbol jerry_port_get_current_time referenced in function ecma_builtin_date_now_helper
ecma-builtin-helpers-date.c.obj : error LNK2019: unresolved external symbol jerry_port_get_local_time_zone_adjustment referenced in function ecma_date_utc
jrt-fatals.c.obj : error LNK2019: unresolved external symbol jerry_port_fatal referenced in function jerry_fatal
bin\jerry-core.dll : fatal error LNK1120: 4 unresolved externals

```
[config-x64-windows-out.log](https://github.com/jerryscript-project/jerryscript/files/7048177/config-x64-windows-out.log)
You can see at https://github.com/microsoft/vcpkg/pull/19743 more details if needed.

rerobika, 2021-12-08:
Hi!

Could you please clarify what do you expect from this issue exactly? Seems like you missing the jerry-port-default lib in pkg-config. Please check the proper usage of pkg-config with JerryScript, here you can find an example https://github.com/jerryscript-project/jerryscript/blob/master/docs/03.API-EXAMPLE.md#before-trying-out-the-examples-get-and-build-jerryscript-library.

autoantwort, 2021-12-09:
Hm seems like that I have to use `tools/build.py` to build the lib and can't use plain CMake