jerryscript-project/jerryscript #4755: Questions: is there any ways to compile to wasm with emscripten?

lee88688, 2021-08-30:
I want to simulate some code with jerryscript using wasm through emscripten. but I found build options without config for emscripten. so there is any ideas to build with emscripten?

akosthekiss, 2021-08-30:
Quite some time ago (in 2016, yikes!), I managed to build JerryScript for emscripten. Here are the notes from those days:

> Environment:
> * OS X 10.11.6 (Darwin 15.6.0)
> * Emscripten 1.35.0 with Node.js v4.1.1 (from: http://kripken.github.io/emscripten-site/docs/getting_started/downloads.html)
> * JerryScript commit 9915307 (Tue Oct 4 11:51:03 2016 +0200)
>
> Commands:
> ```bash
> cmake -H. -Bbuild -DJERRY_LIBC=OFF -DJERRY_LIBM=OFF -DCMAKE_C_COMPILER=emcc -DENABLE_LTO=OFF -DENABLE_ALL_IN_ONE=ON -DENABLE_STRIP=OFF -DCMAKE_EXECUTABLE_SUFFIX_C=.js
> make -C build
> cd build/bin && cat ../../tests/hello.js | node jerry.js --no-prompt
> ```
>
> Note:
> * Only REPL works. Passing JS files as arguments (e.g., `node jerry.js ../../tests/hello.js`) fails with "Error: failed to open file".

Since then, quite some things have changed: sources, build system, emcc, node, ... The above will not work out of the box for sure. I'd start the experiment with the following build configuration:

```bash
rm -rf build/
cmake -H. -Bbuild -DCMAKE_C_COMPILER=emcc -DENABLE_LTO=OFF -DENABLE_AMALGAM=ON -DENABLE_STRIP=OFF -DCMAKE_EXECUTABLE_SUFFIX_C=.js
make -C build VERBOSE=1
```

The above commands get me quite far, but linking fails as I don't have emsdk installed, and the Ubuntu-provided emcc has llvm triplet issues. I hope that someone can pick it up from here.

lee88688, 2021-08-31:
there is any way to make a toolchain to  build? I make a toolchain file and config in command line in build.py, but when I build, it always use windows build tools, here is my toolchain code 
```cmake
set(CMAKE_SYSTEM_NAME emscripten)
set(CMAKE_SYSTEM_PROCESSOR wasm)

set(CMAKE_C_COMPILER emcc)
```

lee88688, 2021-08-31:
and I found any other toolchain in windows don't work either。

akosthekiss, 2021-08-31:
More details are needed to find the root of the problem. How did you invoke `build.py`? What was the output?

Moreover, finding the correct configuration options for an emscripten build is independent from using a toolchain file. So, I suggest to try and configure cmake directly and when the build finally works, move the cmake command line options into a toolchain file.

lee88688, 2021-08-31:
currently I can successfully use emscripten sdk to build lib, here is my code
```bash
emcmake cmake -Bbuild -DJERRY_DEBUGGER=ON -DJERRY_LINE_INFO=ON  -DJERRY_CPOINTER_32_BIT=ON -DJERRY_SYSTEM_ALLOCATOR=ON
emmake make -C build
```
but I have not tested yet.

closed 2021-09-15 by lee88688