jerryscript-project/jerryscript #4767: unknown reference to jerry_init, jerry_cleanup,.....

edword01, 2021-09-14:
I build the jerryscript successfully, in MSYS2, using:
sh''
cmake ../ -DCMAKE=$IDF_PATH/tools/cmake/toolchain-esp32.cmake "-GMSYS Makefiles" -DEXTERNAL_COMPILE_FLAGS="-ffunction-sections -fdata-sections -fstrict-volatile-bitfields" -DJERRY_GLOBAL_HEAP_SIZE=64 -DCMAKE_SYSTEM_NAME=Generic -DCMAKE_SYSTEM_PROCESSOR=xtensa

python tools/build.py --toolchain=cmake/toolchain-esp32.cmake --cmake-param "-GMSYS Makefiles" --jerry-cmdline=OFF --jerry-port-default=ON --lto=OFF --strip=OFF --mem-heap=64
''

Then build the IDF project, and try to use jerry_init() function, I got :

Error:

undefined reference to 'jerry_value_is_error'...
.....
undefined reference to 'jerry_run'
....
How to Fix this error???

zherczeg, 2021-09-14:
This sounds like a linking issue, maybe the `lib*.a` archive was not found, or a wrong one found.

edword01, 2021-09-14:
i am using MSYS2, to compile that jerryscript,  and i think there is some thing wrong with my cmake parammeters,  Could I send u some pictures to show how i compile this,  really need help

galpeter, 2021-09-14:
Please check if you have the `libjerry-core.a` file in your Jerryscript build directory. If the archive is there that is a good start. Additionally the application which uses the jerry API should use this archive file during linking. This is not a jerryscript level configuration but the target application's build options. What is your build/link command for the application using the jerryscript library? The `libjerry-core.a` and `libjerry-port-default.a` should be in the link command.

edword01, 2021-09-14:
This is how i build the jerryscript in MSYS2 in windows:

enter the build directory,  and command:

and then:

cd ../

python tools/build.py --toolchain=cmake/toolchain-esp32.cmake --cmake-param "-GMSYS Makefiles" --jerry-cmdline=OFF --jerry-port-default=ON --lto=OFF --strip=OFF

zherczeg, 2021-09-14:
I have never used msys2, but the general concept is that you get object files during the compilation and they are collected into a library/archive file. Usually there is a tool to check the object files in the library/archive, so you can find where the function you need is exported.

In some environments the order of objects / libraries / archives passed to the linker also matters, so any export needed by an object must be provided by an object / library / archive later in the list passed to the linker.

edword01, 2021-09-14:
When link the js with my idf project, get this error.

but I use nm tool to check that, the libjerry-ext.a has "jerryx_open_handle_scope" function. but I don't known why I get this .

galpeter, 2021-09-14:
for the idf project did you added the jerry archives for the build command (so the idf project is linking program with the jerryscript archives)?

edword01, 2021-09-15:
Now , I have solved this problem, by recompile the jerryscript, and I use the jerryscript files from nodemcujs project, because it already has the component to load js file , and run js file .  Now i can compile success , and it works ok. 
thanks very much.

And now I have another question, how can i pass the array, from the function of xxx.js file  to the C function, and parsing it for  data processing?

rerobika, 2021-12-08:
The last raised question is unrelated to the original issue, please open a new for it if it's still needed.