jerryscript-project/jerryscript #4770: How to get the array from js file?

edword01, 2021-09-16:
I want to use a function in XXX.js file , like this "test_function([0x121212, 0x121212])", 
and I have successfully register the test_function ,  but don't known how acquire this array value.

I didn't find some related function of jerryscript to acquire this parameters from js file?

zherczeg, 2021-09-16:
Please check this file: https://github.com/jerryscript-project/jerryscript/blob/master/docs/02.API-REFERENCE.md

Read the documentation about `jerry_get_property_by_index` and `jerry_get_array_length` functions.
In general I recommend to read this file to understand the API provided by JerryScript. There are big changes since the last release.

galpeter, 2021-09-16:
There is also an api example here: https://github.com/jerryscript-project/jerryscript/blob/master/docs/03.API-EXAMPLE.md#example-5-passing-and-processing-arguments-for-native-c-code  I think this should help to see how the arguments are passed to a C method (form js)

edword01, 2021-09-16:
Thanks, now i successfully pass the parameters, but still a little problem here
In js file, call:
test_function([0xf31, 0xf41, 0xf51])

Then use:
uint16_t buffer[10];
 object_value = jerry_get_property_by_index(args_p[0], 0);
   buffer[0] = jerry_value_to_number(object_value);

then the real value of buffer[0]  i got  is  "0xf310" ,  not "0xf31",  it  looks like bytes left-aligned  form,  how to configure that , to get the right-aligned value  ????

zherczeg, 2021-09-16:
Please read the documentation of the API functions. The return value of jerry_get_property_by_index is a `jerry_value_t`, because it can be an object, string, exception, etc. You can check if it is a number by `jerry_value_is_number` and get its value by `jerry_get_number_value`.

galpeter, 2021-09-16:
Please enable warnings (possibly `-Wall`) in your project, for the line:
```
buffer[0] = jerry_value_to_number(object_value);
```
there (should be) is a conversion warning for this operation:
```
conversion from ‘jerry_value_t’ {aka ‘unsigned int’} to ‘uint16_t’ {aka ‘short unsigned int’} may change value
```

As per API reference documentation, the `jerry_value_to_number` method returns a `jerry_value_t` which is not a valid number in C (it is a JS number value). To extract the value from such `jerry_value_t` (which are numbers) use the previously mentioned `jerry_get_number_value`. Please check these API methods what are the return values for these methods.

edword01, 2021-09-16:
Thanks very much, the jerry_get_number_value fucn works.  And I am going to try : passing an arrya that has 300 elements. hopping that will work well.

Another question,  do Jerryscript  have  event function , or  event listener?  I want to acquire the sensor data(like MPU sensor data) in JS file in real time.,  and the C function driver is already done, what references should I look for ?

zherczeg, 2021-09-17:
JerryScript supports Promise objects and they can be used to subscribe for events provided by your environment. Please read the specification about them: https://262.ecma-international.org/11.0/#sec-promise-objects

You can provide promises to the ECMAScript world, and resolve them using the JerryScript API (e.g. `jerry_create_promise`). To run the Promise `then` handlers or `async` functions, use `jerry_run_all_enqueued_jobs`.

kimyLee, 2021-09-23:
> JerryScript supports Promise objects and they can be used to subscribe for events provided by your environment. Please read the specification about them: [...]

@zherczeg , an enlightening way for me ! by the ways, what about the event listener? in the above case,  ECMAScript world can use `then` function of the provided promises, however, we know the `promise`  can be resolved only once. How can I do something like the `click event listener`? That means I can listen the event in the ECMAScript world , and when every click event happen on device, the C function can call `jerry_xxx` to carry the event listener in the ECMAScript world, not just once call. Does anyway make this?

kimyLee, 2021-09-23:
> jerry_create_promise

I just come up a idea, that I new I proxy use `jerry_create_proxy `, and provide to the ECMAScript world, where I can defined its set function, then when I set the value in the C function , it should get the value in the ECMAScript world ?

zherczeg, 2021-09-23:
If you want an `onclick` like chain resolving, you can use async generators. It generates a promise for the next click, and when it is resolved by `then`, the handler function can ask another promise. Specification:
https://262.ecma-international.org/11.0/#sec-async-generator-function-definitions

Of course you can use only the concept, A simplified example with a C external function:
```
createPromiseForTheNextClickExternalFunction().then(function selfRef(event) {
  // Do something using the event data
  // Create a promise for the next click event
  createPromiseForTheNextClickExternalFunction().then(selfRef)
})
```
Your system needs to remember only a single promise, which is resolved when the event happens (then it throws away and create a new promise for the next event). Of course multiple listeners can attach to the same promise, if more function needs to be called when an event occures.

zherczeg, 2021-09-23:
Async functions can make this even simpler:
```
(async function() {
  // Exit from the loop if you don't want more events
  while (true) {
    var event = await createPromiseForTheNextClickExternalFunction()
    // use event
  }
})()

kimyLee, 2021-09-23:
> Async functions can make this even simpler:
> 
> ```
> (async function() {
>   // Exit from the loop if you don't want more events
>   while (true) {
>     var event = await createPromiseForTheNextClickExternalFunction()
>     // use event
>   }
> })()
> ```

3s very much, I will take a try soon