jerryscript-project/jerryscript #4779: stack-overflow in jmem_heap_alloc_block_internal

marckwei, 2021-09-23:
###### JerryScript revision

e1ce7dd7

###### Build platform

Ubuntu 20.04.3 LTS (Linux 5.11.0-34-generic x86_64)
Linux-5.11.0-34-generic-x86_64-with-glibc2.29

###### Build steps

./tools/build.py --clean --debug --compile-flag=-fsanitize=address \
--compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer \
--compile-flag=-fno-common --compile-flag=-g --strip=off \
--system-allocator=on --logging=on --linker-flag=-fuse-ld=gold \
--error-messages=on --lto=off --stack-limit 10

###### Test case

```javascript
function opt() {
    let obj = new opt();
    for (let i = 0; i < 1; i++) {
        obj.x = 1;
        obj = +obj;
        obj.x = 1;
    }
}

function main() {
    for (let i = 0; i < 100; i++) {
        opt();
    }
}

main();
```

###### Execution platform

the same as the build platform.

###### Output

```
AddressSanitizer:DEADLYSIGNAL
=================================================================
==1259593==ERROR: AddressSanitizer: stack-overflow on address 0xff275f1c (pc 0xf7a73bdb bp 0xff276348 sp 0xff275f00 T0)
    #0 0xf7a73bda in __interceptor_malloc (/lib32/libasan.so.5+0x113bda)
    #1 0x5661e74a in jmem_heap_alloc_block_internal (/home/weiwei/Desktop/jerry+0x3774a)
    #2 0x56620ba1 in jmem_pools_alloc /home/weiwei/Desktop/jerryscript/jerry-core/jmem/jmem-poolman.c:108
    #3 0x5681956c in ecma_alloc_object /home/weiwei/Desktop/jerryscript/jerry-core/ecma/base/ecma-alloc.c:81
    #4 0x56697174 in ecma_create_object /home/weiwei/Desktop/jerryscript/jerry-core/ecma/base/ecma-helpers.c:85
    #5 0x56707bdc in ecma_op_function_construct /home/weiwei/Desktop/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1558
    #6 0x56815ac8 in opfunc_construct /home/weiwei/Desktop/jerryscript/jerry-core/vm/vm.c:884
    #7 0x56815ac8 in vm_execute /home/weiwei/Desktop/jerryscript/jerry-core/vm/vm.c:5258
    #8 0x568190a3 in vm_run /home/weiwei/Desktop/jerryscript/jerry-core/vm/vm.c:5338
    #9 0x56701549 in ecma_op_function_call_simple /home/weiwei/Desktop/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1131
    #10 0x56707c47 in ecma_op_function_construct /home/weiwei/Desktop/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1573
    #11 0x56815ac8 in opfunc_construct /home/weiwei/Desktop/jerryscript/jerry-core/vm/vm.c:884
    #12 0x56815ac8 in vm_execute /home/weiwei/Desktop/jerryscript/jerry-core/vm/vm.c:5258
    #13 0x568190a3 in vm_run /home/weiwei/Desktop/jerryscript/jerry-core/vm/vm.c:5338
    #14 0x56701549 in ecma_op_function_call_simple /home/weiwei/Desktop/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1131
    #15 0x56707c47 in ecma_op_function_construct /home/weiwei/Desktop/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1573
    #16 0x56815ac8 in opfunc_construct /home/weiwei/Desktop/jerryscript/jerry-core/vm/vm.c:884
    #17 0x56815ac8 in vm_execute /home/weiwei/Desktop/jerryscript/jerry-core/vm/vm.c:5258
    #18 0x568190a3 in vm_run /home/weiwei/Desktop/jerryscript/jerry-core/vm/vm.c:5338
    #19 0x56701549 in ecma_op_function_call_simple /home/weiwei/Desktop/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1131
    #20 0x56707c47 in ecma_op_function_construct /home/weiwei/Desktop/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1573
    #21 0x56815ac8 in opfunc_construct /home/weiwei/Desktop/jerryscript/jerry-core/vm/vm.c:884
[... 270 lines cut ...]
    #292 0x56815ac8 in vm_execute /home/weiwei/Desktop/jerryscript/jerry-core/vm/vm.c:5258
    #293 0x568190a3 in vm_run /home/weiwei/Desktop/jerryscript/jerry-core/vm/vm.c:5338
    #294 0x56701549 in ecma_op_function_call_simple /home/weiwei/Desktop/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1131
    #295 0x56707c47 in ecma_op_function_construct /home/weiwei/Desktop/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1573
    #296 0x56815ac8 in opfunc_construct /home/weiwei/Desktop/jerryscript/jerry-core/vm/vm.c:884
    #297 0x56815ac8 in vm_execute /home/weiwei/Desktop/jerryscript/jerry-core/vm/vm.c:5258
    #298 0x568190a3 in vm_run /home/weiwei/Desktop/jerryscript/jerry-core/vm/vm.c:5338
    #299 0x56701549 in ecma_op_function_call_simple /home/weiwei/Desktop/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1131
    #300 0x56707c47 in ecma_op_function_construct /home/weiwei/Desktop/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1573
    #301 0x56815ac8 in opfunc_construct /home/weiwei/Desktop/jerryscript/jerry-core/vm/vm.c:884
    #302 0x56815ac8 in vm_execute /home/weiwei/Desktop/jerryscript/jerry-core/vm/vm.c:5258
    #303 0x568190a3 in vm_run /home/weiwei/Desktop/jerryscript/jerry-core/vm/vm.c:5338
    #304 0x56701549 in ecma_op_function_call_simple /home/weiwei/Desktop/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1131
    #305 0x56707c47 in ecma_op_function_construct /home/weiwei/Desktop/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1573
    #306 0x56815ac8 in opfunc_construct /home/weiwei/Desktop/jerryscript/jerry-core/vm/vm.c:884
    #307 0x56815ac8 in vm_execute /home/weiwei/Desktop/jerryscript/jerry-core/vm/vm.c:5258

SUMMARY: AddressSanitizer: stack-overflow (/lib32/libasan.so.5+0x113bda) in __interceptor_malloc
==1259593==ABORTING

```

###### Backtrace

see above

zherczeg, 2021-09-23:
`new opt()` is a recursive call. Enable stack check build option to avoid this. Your code does not make much sense to me, what are you trying to do?

marckwei, 2021-09-25:
> `new opt()` is a recursive call. Enable stack check build option to avoid this. Your code does not make much sense to me, what are you trying to do?

This sample is just for testing. `--stack-limit 10` was applied during the build,but the program still crashed.

zherczeg, 2021-09-25:
This is obviously an inifinite recursive call, maybe a stack limit check could be added somewhere to stop this. Do you want to fix it? This is a great opportunity for you to learn about the engine internals.

rerobika, 2021-09-25:
I think you should simply reduce the limit for e.g.5, 2, or 1 KB