jerryscript-project/jerryscript #4781: Unfreed float causing memory leak in ecma-regexp-object

Symatrix6, 2021-09-27:
###### JerryScript revision
14ff5bf

###### Build platform
Ubuntu 20.04.3 LTS (Linux 5.11.0-34-generic x86_64)

###### Build steps
```sh
./tools/build.py --profile=es2015-subset --lto=off --compile-flag=-g \
--clean --debug --strip=off --logging=on --error-messages=on \
--compile-flag=-fsanitize=address --stack-limit=20
```

###### Test case
```javascript
function testAdvanceStringIndex(lastIndex) {
  let exec_count = 0;
  let last_last_index = -1;
  let fake_re = {
    exec: () => {
      return exec_count++ == 0 ? [""] : null;
    },

    get lastIndex() {
      return lastIndex;
    },

    set lastIndex(value) {
    },

    get global() {
      return true;
    }
  };

  RegExp.prototype[Symbol.match].call(fake_re, "abc");
}

testAdvanceStringIndex(0x7ffffff);
```

###### Output
```sh
ICE: Assertion 'JERRY_CONTEXT (jmem_heap_allocated_size) == 0' failed at /jerryscript/jerry-core/jmem/jmem-heap.c(jmem_heap_finalize):107.
Error: ERR_FAILED_INTERNAL_ASSERTION
Aborted (core dumped)
```

###### Backtrace
```sh
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007f45b1399859 in __GI_abort () at abort.c:79
#2  0x00005574cdd6736f in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at /jerryscript/jerry-port/default/default-fatal.c:30
#3  0x00005574cdcff841 in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at /jerryscript/jerry-core/jrt/jrt-fatals.c:63
#4  0x00005574cdcff897 in jerry_assert_fail (assertion=0x5574cdd73de8 "JERRY_CONTEXT (jmem_heap_allocated_size) == 0", file=0x5574cdd73d60 "/jerryscript/jerry-core/jmem/jmem-heap.c", function=0x5574cdd8bce0 <__func__.6665> "jmem_heap_finalize", line=107) at /jerryscript/jerry-core/jrt/jrt-fatals.c:87
#5  0x00005574cdcfe8a0 in jmem_heap_finalize () at /jerryscript/jerry-core/jmem/jmem-heap.c:107
#6  0x00005574cdcfe5d7 in jmem_finalize () at /jerryscript/jerry-core/jmem/jmem-allocator.c:170
#7  0x00005574cdca9aaf in jerry_cleanup () at /jerryscript/jerry-core/api/jerry.c:232
#8  0x00005574cdca73b4 in main (argc=2, argv=0x7ffffff6d468) at /jerryscript/jerry-main/main-jerry.c:371
#9  0x00007f45b139b0b3 in __libc_start_main (main=0x5574cdca6889 <main>, argc=2, argv=0x7ffffff6d468, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7ffffff6d458) at ../csu/libc-start.c:308
#10 0x00005574cdca67ce in _start ()
```

###### Expected behavior
According to our analysis, the root cause of this assertion failed is at /jerryscript/jerry-core/ecma/ecma-regexp-object.c:3497. While getting next_set_status, function ecma-op-object-put() called function ecma_make_length_value(index), which alloc a 8-bit chunk memory use as a float number if index is larger than 0x7ffffff. This chunk is not freed, causing assertion failed. To repair, ecma_make_length_value(index) should be replaced by last_index created by ecma-regexp-object.c:3496.

closed 2021-10-01 by zherczeg

Symatrix6, 2021-10-02:
Hi there, we want to apply for a CVE number, could you please help us?

akosthekiss, 2021-10-04:
@Symatrix6 We are not a CNA. Please, consult https://cve.mitre.org/cve/request_id.html

PR #4787 (2021-10-01): Fix leaking lastIndex values in RegExp built-ins
Fixes #4781. This is a followup fix after #4166.

JerryScript-DCO-1.0-Signed-off-by: Csaba Osztrogonác csaba.osztrogonac@h-lab.eu
commit f3cd5860 (2021-10-01): Fix leaking lastIndex values in RegExp built-ins (#4787)

Fixes #4781. This is a followup fix after #4166.

JerryScript-DCO-1.0-Signed-off-by: Csaba Osztrogonác csaba.osztrogonac@h-lab.eu