jerryscript-project/jerryscript #4782: How to use promise in jerryscript?

edword01, 2021-09-28:
when create promise in C file, like this:
E.g.:
```sh
jerry_value_t test_promise = jerry_create_promise();
jerry_value_t argument = jerry_create_string((jerry_char_t*)"promise");
 jerry_value_t is_ok = jerry_resolve_or_reject_promise (test_promise,
                                                                    argument,
                                                                    true);
 ```
and then,  how to get this promise in js?? where to find an example?

zherczeg, 2021-09-28:
There are many ways to pass any value to ES world. You can use it for any value, not just promises. Some examples:

- Return the value from an external function: https://github.com/jerryscript-project/jerryscript/blob/master/docs/02.API-REFERENCE.md#jerry_create_external_function
- Get a callback, and call it with https://github.com/jerryscript-project/jerryscript/blob/master/docs/02.API-REFERENCE.md#jerry_call_function
- Add it as a property for an existing object (global object, an object passed to an external callback, etc.) https://github.com/jerryscript-project/jerryscript/blob/master/docs/02.API-REFERENCE.md#jerry_set_property

Don't forget about reference counting.

edword01, 2021-09-28:
Not only pass value to ES world,  I want to get the sensor value from C ,  and do this job in JS like event listener.  so Promise object can use Promise.then to handle this .
now i set several task in RTOS,  

TaskA:
    call jerry_init,  mount_spifff,  .... .......  ,,  but doesn't  create promise object.

TaskB:
    call  func  jerry_create_promise  ,   and  jerry_resolve_or_reject_promise,  like above,   in every 3seconds. 

now  in  JS file , how to quote this promise obj  create in C ?  Don't know the method  or gramma  for jerryscript

zherczeg, 2021-09-28:
Don't forget JerryScript is not thread safe, so accessing the API from multiple threads at the same time will likely lead to crash.

Your workflow should look like this:

- Have a JerryScript thread, which runs your ES code
- Create a "current promise" which will be resolved when the event occures
- Create an external function, which returns with this "current promise" (don't forget to increase its refcount). The ES code can call this external function to get the "current promise"
- When the event occures, notify the JerryScript thread
- The JerryScript thread resolves the "current promise", release it and creates a new current one (this step can be delayed until the external function is actually called)
- Don't forget to run the jobqueue, otherwise promises don't work

edword01, 2021-09-28:
what's the problem of this workflow:
in rtos, have a jerryscript task,  create external func "get_promise",  create promise,  load js file,  then come into a while loop to wait a Semaphore ,  Semaphore is gived by another task,  if it is gived, do the job of "jerry_resolve_or_reject_promise (promise_obj, argument, true);  jerry_release_value(promise_obj);  promise_obj = jerry_create_promise();"

loop like:

while(1)
{
    .......
     if(waitting the semaphore 
     {
      //resolve promise, and create new one ;
     }

    job_value = jerry_run_all_enqueued_jobs ();
  //check  job_value  
  .....
}

and in the external functio "get_promise",  just "return promise_obj",

In js file,  var p = get_promise();
try {
    //console.log('global_test', global_test)
    // if (promise_test.then) {
    //     console.log('promise_test', promise_test.then)
    // }
    p.then(function(x){
        //console.log("%s", x);
        console.log("promise");
    });
}

All this crash now.

zherczeg, 2021-09-28:
Your approach is correct. The crash might be caused by reference counting (e.g. not calling `jerry_acquire_value` in your `get_promise` external function). The `jerry_resolve_or_reject_promise` rarely throws an error, so the value passed to it might be invalid as well (e.g. a value which was freed)

edword01, 2021-09-28:
So in get_promise func, use this?
####
jerry_value_t value = jerry_acquire_value(test_promise);
return value;

in js file, promise still doesn't work.

zherczeg, 2021-09-28:
yes, `return jerry_acquire_value(test_promise)`returns with another reference to the promise. There might be other errors in the code though.

Sherry0901, 2021-12-08:
Have you solved the problem？ @edword01