jerryscript-project/jerryscript #4864: Is it possible to implement worker in JerryScript?

Sherry0901, 2021-12-08:
Is it possible to implement worker(like web worker) in JerryScript?
if it's available, where can I get some examples or guides?

rerobika, 2021-12-08:
Hi, please check the following example: https://github.com/jerryscript-project/jerryscript/blob/master/docs/02.API-REFERENCE.md#jerry_context_alloc

Sherry0901, 2021-12-09:
thanks!
I see, so JerryScript can create multi-threads.
But I am still confused about the details.
How can I implement postMessage,addEventListener,removeEventListener...in JerryScript? @rerobika

rerobika, 2021-12-09:
> I see, so JerryScript can create multi-threads.
Nope, JerryScript cannot create threads. The linked example only shows that the engine is able to work in a multi threaded context. Please note that those independent contexts cannot pass value to each other. 

If I were you I'd definitely use [Promises](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Promise) or [async functions](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Statements/async_function) to build and own event queue to handle the listener events. However, it's still single threaded, but the jobs can communicate with each other.