jerryscript-project/jerryscript #4865: AddressSanitizer: stack-overflow jerry-core/vm/vm.c:975 in vm_loop

eternalsakura, 2021-12-08:
## JerryScript commit hash
55acdf2048b390d0f56f12e64dbfb2559f0e70ad

## Build platform
Ubuntu 20.04 LTS

## Build steps
```
python3 tools/build.py --debug --compile-flag=-fsanitize=address
```
## poc
```
unction assertThrows(code, type_opt, cause_opt) {
    if (typeof code === 'function')
        return code();
    if (typeof code === 'string')
        return eval(code);
}
var global = this;
global.__defineSetter__('x', function (v) {
    x = v;
});
assertThrows('global.x = 0', RangeError);
```

## asan log
```
AddressSanitizer:DEADLYSIGNAL
=================================================================
==503383==ERROR: AddressSanitizer: stack-overflow on address 0x7fff9e2c6e28 (pc 0x558b17c76d5a bp 0x7fff9e2c75c0 sp 0x7fff9e2c6e10 T0)
    #0 0x558b17c76d59 in vm_loop /home/sakura/jerryscript/jerry-core/vm/vm.c:975
    #1 0x558b17c93b85 in vm_execute /home/sakura/jerryscript/jerry-core/vm/vm.c:5260
    #2 0x558b17c9420b in vm_run /home/sakura/jerryscript/jerry-core/vm/vm.c:5363
    #3 0x558b17bf620f in ecma_op_function_call_simple /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1203
    #4 0x558b17bf6d93 in ecma_op_function_call /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1439
    #5 0x558b17c06014 in ecma_op_object_put_with_receiver /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-objects.c:1677
    #6 0x558b17c04680 in ecma_op_object_put /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-objects.c:1197
    #7 0x558b17cd378e in ecma_op_put_value_lex_env_base /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-get-put-value.c:293
    #8 0x558b17c913e8 in vm_loop /home/sakura/jerryscript/jerry-core/vm/vm.c:4751
    #9 0x558b17c93b85 in vm_execute /home/sakura/jerryscript/jerry-core/vm/vm.c:5260
    #10 0x558b17c9420b in vm_run /home/sakura/jerryscript/jerry-core/vm/vm.c:5363
    #11 0x558b17bf620f in ecma_op_function_call_simple /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1203
    #12 0x558b17bf6d93 in ecma_op_function_call /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1439
    #13 0x558b17c06014 in ecma_op_object_put_with_receiver /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-objects.c:1677
    #14 0x558b17c04680 in ecma_op_object_put /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-objects.c:1197
    #15 0x558b17cd378e in ecma_op_put_value_lex_env_base /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-get-put-value.c:293
    #16 0x558b17c913e8 in vm_loop /home/sakura/jerryscript/jerry-core/vm/vm.c:4751
    #17 0x558b17c93b85 in vm_execute /home/sakura/jerryscript/jerry-core/vm/vm.c:5260
    #18 0x558b17c9420b in vm_run /home/sakura/jerryscript/jerry-core/vm/vm.c:5363
    #19 0x558b17bf620f in ecma_op_function_call_simple /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1203
    #20 0x558b17bf6d93 in ecma_op_function_call /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1439
    #21 0x558b17c06014 in ecma_op_object_put_with_receiver /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-objects.c:1677
    #22 0x558b17c04680 in ecma_op_object_put /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-objects.c:1197
    #23 0x558b17cd378e in ecma_op_put_value_lex_env_base /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-get-put-value.c:293
    #24 0x558b17c913e8 in vm_loop /home/sakura/jerryscript/jerry-core/vm/vm.c:4751
    #25 0x558b17c93b85 in vm_execute /home/sakura/jerryscript/jerry-core/vm/vm.c:5260
    #26 0x558b17c9420b in vm_run /home/sakura/jerryscript/jerry-core/vm/vm.c:5363
    #27 0x558b17bf620f in ecma_op_function_call_simple /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1203
    #28 0x558b17bf6d93 in ecma_op_function_call /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1439
    #29 0x558b17c06014 in ecma_op_object_put_with_receiver /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-objects.c:1677
    #30 0x558b17c04680 in ecma_op_object_put /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-objects.c:1197
    #31 0x558b17cd378e in ecma_op_put_value_lex_env_base /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-get-put-value.c:293
    #32 0x558b17c913e8 in vm_loop /home/sakura/jerryscript/jerry-core/vm/vm.c:4751
    #33 0x558b17c93b85 in vm_execute /home/sakura/jerryscript/jerry-core/vm/vm.c:5260
    #34 0x558b17c9420b in vm_run /home/sakura/jerryscript/jerry-core/vm/vm.c:5363
    #35 0x558b17bf620f in ecma_op_function_call_simple /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1203
    #36 0x558b17bf6d93 in ecma_op_function_call /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1439
    #37 0x558b17c06014 in ecma_op_object_put_with_receiver /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-objects.c:1677
    #38 0x558b17c04680 in ecma_op_object_put /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-objects.c:1197
    #39 0x558b17cd378e in ecma_op_put_value_lex_env_base /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-get-put-value.c:293
[... 187 lines cut ...]
    #227 0x558b17bf620f in ecma_op_function_call_simple /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1203
    #228 0x558b17bf6d93 in ecma_op_function_call /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1439
    #229 0x558b17c06014 in ecma_op_object_put_with_receiver /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-objects.c:1677
    #230 0x558b17c04680 in ecma_op_object_put /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-objects.c:1197
    #231 0x558b17cd378e in ecma_op_put_value_lex_env_base /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-get-put-value.c:293
    #232 0x558b17c913e8 in vm_loop /home/sakura/jerryscript/jerry-core/vm/vm.c:4751
    #233 0x558b17c93b85 in vm_execute /home/sakura/jerryscript/jerry-core/vm/vm.c:5260
    #234 0x558b17c9420b in vm_run /home/sakura/jerryscript/jerry-core/vm/vm.c:5363
    #235 0x558b17bf620f in ecma_op_function_call_simple /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1203
    #236 0x558b17bf6d93 in ecma_op_function_call /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1439
    #237 0x558b17c06014 in ecma_op_object_put_with_receiver /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-objects.c:1677
    #238 0x558b17c04680 in ecma_op_object_put /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-objects.c:1197
    #239 0x558b17cd378e in ecma_op_put_value_lex_env_base /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-get-put-value.c:293
    #240 0x558b17c913e8 in vm_loop /home/sakura/jerryscript/jerry-core/vm/vm.c:4751
    #241 0x558b17c93b85 in vm_execute /home/sakura/jerryscript/jerry-core/vm/vm.c:5260
    #242 0x558b17c9420b in vm_run /home/sakura/jerryscript/jerry-core/vm/vm.c:5363
    #243 0x558b17bf620f in ecma_op_function_call_simple /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1203
    #244 0x558b17bf6d93 in ecma_op_function_call /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1439
    #245 0x558b17c06014 in ecma_op_object_put_with_receiver /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-objects.c:1677
    #246 0x558b17c04680 in ecma_op_object_put /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-objects.c:1197
    #247 0x558b17cd378e in ecma_op_put_value_lex_env_base /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-get-put-value.c:293
    #248 0x558b17c913e8 in vm_loop /home/sakura/jerryscript/jerry-core/vm/vm.c:4751

SUMMARY: AddressSanitizer: stack-overflow /home/sakura/jerryscript/jerry-core/vm/vm.c:975 in vm_loop
```

eternalsakura, 2021-12-08:
- poc2
```
(function () {
    try {
        Function("this.__defineGetter__(\"x\",(Function(\"for(z=0;z<6;z++)(x)\")))")();
    } catch (e) {
    }
}()(function f(d, aaaaaa) {
    if (bbbbbb = aaaaaa) {
        x;
    }
    f(bbbbbb, aaaaaa + 1);
}([], 0)));
```

rerobika, 2021-12-08:
Closing the issue, see #4866 comment.

closed 2021-12-08 by rerobika