jerryscript-project/jerryscript #4874: Assertion '!is_static' failed at jerry-core/parser/js/js-parser-expr.c(parser_parse_class_body):729

eternalsakura, 2021-12-09:
## JerryScript commit hash
55acdf2048b390d0f56f12e64dbfb2559f0e70ad

## Build platform
Ubuntu 20.04 LTS

## Build steps
```
./tools/build.py --clean --debug --compile-flag=-fsanitize=address --compile-flag=-m32 --lto=off --logging=on --line-info=on --error-message=on --system-allocator=on --profile=es2015-subset --stack-limit=20
```

## poc
```
function assertSyntaxError(code) {
    try {
        eval(code);
        throw new Error('Should throw SyntaxError, but executed code without throwing');
    } catch (e) {
        if (!e instanceof SyntaxError)
            throw new Error('Should throw SyntaxError, but threw ' + e);
    }
}
assertSyntaxError('let C = class { #constructor() {} }');
assertSyntaxError('let C = class { static #constructor() {} }');
assertSyntaxError('class C { #constructor() {} }');
assertSyntaxError('class C { static #constructor() {} }');
assertSyntaxError('let C = class { get #constructor() {} }');
assertSyntaxError('let C = class { static get #constructor() {} }');
assertSyntaxError('class C { get #constructor() {} }');
assertSyntaxError('class C { static get #constructor() {} }');
assertSyntaxError('let C = class { set #constructor(v) {} }');
assertSyntaxError('let C = class { static set #constructor(v) {} }');
assertSyntaxError('class C { set #constructor(v) {} }');
assertSyntaxError('class C { static set #constructor(v) {} }');
assertSyntaxError('let C = class { #constructor; }');
assertSyntaxError('let C = class { static #constructor; }');
assertSyntaxError('class C { #constructor; }');
assertSyntaxError('class C { static #constructor; }');
```

## assert log
```
ICE: Assertion '!is_static' failed at /home/sakura/jerryscript/jerry-core/parser/js/js-parser-expr.c(parser_parse_class_body):729.
Error: ERR_FAILED_INTERNAL_ASSERTION
[1]    4022294 abort      ~/jerryscript/build2/bin/jerry fuzz_output/fuzzer1/.cur_input
```

eternalsakura, 2021-12-09:
I have a little doubt about --system-allocator=on, should I enable it?
Do you have a suitable configuration for fuzzing? thanks

eternalsakura, 2021-12-09:
To prevent me from using the wrong configuration and submitting an incorrect issue

eternalsakura, 2021-12-09:
For example, I have a problem now. When --system-allocator=on and stack-limit is 20, I can trigger a stack overflow, but this stack overflow cannot trigger it when system-allocator=off .

In the same way, I have another poc, the situation is just the opposite, so I am here to ask for your suggestions, which configuration poc do you accept

rerobika, 2021-12-09:
Hi!

I suggest to use the following config:

```
./tools/build.py --clean --debug --compile-flag=-fsanitize=address \
--compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer \
--compile-flag=-fno-common --compile-flag=-g \
--strip=off --system-allocator=on --logging=on \
--linker-flag=-fuse-ld=gold --error-messages=on --line-info=ON \
--stack-limit=10
```

PR #4880 (2021-12-15): Fix class constructor as accessor check
This patch fixes #4874.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik robert.fancsik@h-lab.eu
commit d37ff33d (2021-12-15): Fix class constructor as accessor check (#4880)

This patch fixes #4874.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik robert.fancsik@h-lab.eu
closed 2021-12-15 by ossy-szeged