jerryscript-project/jerryscript #4893: Assertion 'JERRY_CONTEXT (jmem_heap_allocated_size) == 0' failed at jmem-heap.c(jmem_heap_finalize).

hope-fly, 2021-12-09:
###### JerryScript revision

Commit: [51da1551](https://github.com/jerryscript-project/jerryscript/commit/51da15516e1436184be9914314f76d596f14cd27)

Version: v3.0.0

###### Build platform

Ubuntu 18.04.5 LTS (Linux 5.4.0-44-generic x86_64)

###### Build steps

```bash
./tools/build.py --clean --debug --profile=es2015-subset --compile-flag=-fsanitize=address --compile-flag=-m32 --lto=off --logging=on --line-info=on --error-message=on --system-allocator=on --stack-limit=20
```

###### Test case

```javascript
let args = new Array(0x10000);
args = args.fill(0x1234).join(', ');
RegExp('new Array(' + args + ')');
```

​	

###### Execution steps & Output

```bash
$ ./jerryscript/build/bin/jerry poc.js

ICE: Assertion 'JERRY_CONTEXT (jmem_heap_allocated_size) == 0' failed at /root/jerryscript/jerry-core/jmem/jmem-heap.c(jmem_heap_finalize):108.
Error: ERR_FAILED_INTERNAL_ASSERTION
```

rerobika, 2021-12-09:
I think we can categorize this test OOM like we discussed in https://github.com/jerryscript-project/jerryscript/issues/4897

closed 2021-12-09 by rerobika

hope-fly, 2021-12-09:
There may be some difference from #4897 ，but I’m not for sure.

rerobika, 2021-12-09:
I've tested with the build command above and I get OOM.

hope-fly, 2021-12-09:
ok, thanks a lot！