jerryscript-project/jerryscript #4895: Assertion 'ecma_object_check_class_name_is_object (obj_p)' failed at ecma-objects.c(ecma_object_get_class_name)

hope-fly, 2021-12-09:
###### JerryScript revision

Commit: [51da1551](https://github.com/jerryscript-project/jerryscript/commit/51da15516e1436184be9914314f76d596f14cd27)

Version: v3.0.0

###### Build platform

Ubuntu 18.04.5 LTS (Linux 5.4.0-44-generic x86_64)

###### Build steps

```bash
./tools/build.py --clean --debug --profile=es2015-subset --compile-flag=-fsanitize=address --compile-flag=-m32 --lto=off --logging=on --line-info=on --error-message=on --system-allocator=on --stack-limit=20
```

###### Test case

```javascript

```

​	

###### Execution steps & Output

```bash
$ ./jerryscript/build/bin/jerry poc.js

ICE: Assertion 'ecma_object_check_class_name_is_object (obj_p)' failed at /root/jerryscript/jerry-core/ecma/operations/ecma-objects.c(ecma_object_get_class_name):3009.
Error: ERR_FAILED_INTERNAL_ASSERTION
```

rerobika, 2021-12-09:
Please, no one is in a hurry...
Before submitting an issue, check:
 - whether the issue template is correctly filled.
 - the issue is reproducible on the **latest** master
 - At least there is no already reported assertion with the same reason.
 
@eternalsakura Please try to follow this short guide above as well.

rerobika, 2021-12-09:
Duplicate of #4884.

closed 2021-12-09 by rerobika

hope-fly, 2021-12-09:
Got it! Thanks for your devotion.