jerryscript-project/jerryscript #4897: Some questions in the fuzz test, hope to be answered, about"AddressSanitizer: allocator is out of memory"

eternalsakura, 2021-12-09:
## JerryScript commit hash
55acdf2048b390d0f56f12e64dbfb2559f0e70ad

## Build platform
Ubuntu 20.04 LTS

## Build steps
```
./tools/build.py --clean --debug --compile-flag=-fsanitize=address \
--compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer \
--compile-flag=-fno-common --compile-flag=-g \
--strip=off --system-allocator=on --logging=on \
--linker-flag=-fuse-ld=gold --error-messages=on --line-info=ON \
--stack-limit=10
```

## some poc
- poc1
```
function shouldThrow(func, errorMessage) {
    let errorThrown = false;
    try {
        func();
    } catch (error) {
        errorThrown = true;
        if (String(error) !== errorMessage)
            throw new Error(`Bad error: ${ error }`);
    }
    if (!errorThrown)
        throw new Error('Not thrown!');
}
const obj = {};
const s = [0.1].toLocaleString().padEnd(2 ** 31 - 1, 'ab');
Object.defineProperty(obj, 'foo', {
    value: s,
    writable: false,
    enumerable: true,
    configurable: false
});
shouldThrow(() => {
    Object.defineProperty(obj, 'foo', {
        value: 'bar'
    });
}, 'TypeError: Attempting to change value of a readonly property.');
```

- poc2
```
var s = 'ሴ-------';
for (var i = 0; i < 17; i++) {
    s += s;
}
s.replace(/[\u1234]/g, '');
```

## asan log
```
==1822485==ERROR: AddressSanitizer: allocator is out of memory trying to allocate 0x1cf60 bytes
==1822485==FATAL: AddressSanitizer: internal allocator is out of memory trying to allocate 0xf bytes
```
```
➜  jerryscript_workdir git:(main)  ASAN_OPTIONS=allocator_may_return_null=1  ~/jerryscript/build2/bin/jerry oom1.js
AddressSanitizer:DEADLYSIGNAL
=================================================================
==1822664==ERROR: AddressSanitizer: SEGV on unknown address 0x00000000 (pc 0x56615065 bp 0xfff7e0a8 sp 0xfff7e080 T0)
==1822664==The signal is caused by a WRITE memory access.
==1822664==Hint: address points to the zero page.
==1822664==FATAL: AddressSanitizer: internal allocator is out of memory trying to allocate 0xf bytes
```

## Question 
Question : Should I report these problems to you as null pointers, or how to prevent them from asan crashing

eternalsakura, 2021-12-09:
In fact, I don’t want to report these problems to you as null pointers, because there are too many, so I would like to suggest that you can give a general fix :)

rerobika, 2021-12-09:
In these cases the engine has `ERR_OUT_OF_MEMORY` exit code. At this point the engine calls `abort` which crashes the ASAN. To resolve this we were talking about to change the OOM behavior to similar to exceptions, but this approach requires a huge rework. So until it's not done, please just ignore the OOM failures.

eternalsakura, 2021-12-09:
Can you point out the specific code, I will use exit(0) instead of abort when fuzzing

thank you very much

eternalsakura, 2021-12-09:
I think abort is not a wrong behavior during oom, and it does not need to be changed to an exception. :)

rerobika, 2021-12-09:
Sure, https://github.com/jerryscript-project/jerryscript/blob/master/jerry-core/include/jerryscript-port.h#L47.

I've just mention that we had a plan like that, since it kinda relates to fuzzing. The main approach is to create the OOM handling more convenient. Catching an exception is more user friendly than catching an abort signal.

eternalsakura, 2021-12-09:
Yes, it is true, thanks for your reply

closed 2021-12-09 by eternalsakura