jerryscript-project/jerryscript #4902: Assertion 'ecma_is_lexical_environment (object_p)' failed at ecma-helpers.c (ecma_get_lex_env_type).

hope-fly, 2021-12-13:
###### JerryScript revision

Commit: [42523bd6](https://github.com/jerryscript-project/jerryscript/commit/42523bd6e2b114755498c9f68fd78545f9b33476)
Version: v3.0.0

###### Build platform

Ubuntu 18.04.5 LTS (Linux 5.4.0-44-generic x86_64)

###### Build steps

```bash
python  ./tools/build.py --clean --debug --compile-flag=-fsanitize=address --compile-flag=-m32 --compile-flag=-fno-omit-frame-pointer --compile-flag=-fno-common --compile-flag=-g --strip=off --system-allocator=on --logging=on --linker-flag=-fuse-ld=gold --error-messages=on --line-info=on --stack-limit=10
```

###### Test case

```javascript
function echo(str) {
  console.log(str);
}

function T(p, r, u) {
  return Object.assign(p, {
    then(onFulfilled, onRejected) {
      if (u) {
        onFulfilled(r);
      } else {
        onFulfilled();
      }

      return Promise.prototype.then.call(this, onFulfilled, onRejected);
    }
  });
}

function JSEtest(i) {
  var ps = [T(Promise.resolve('success'))];
  Promise.all(ps).then(res => {
    echo(`Test #${i} - Success with '${res}' (length = ${res.length}) (isArray = ${Array.isArray(res)})`);
  }).catch(err => {
    echo(`Test #${i} - Catch with ${err}`);
  });
}

JSEtest(1);
```

###### Execution steps & Output

```bash
$ ./jerryscript/build/bin/jerry poc.js

ICE: Assertion 'ptr != NULL' failed at /root/jerryscript/jerry-core/ecma/base/ecma-helpers-value.c(ecma_get_pointer_from_ecma_value):115.
Error: ERR_FAILED_INTERNAL_ASSERTION
```

hope-fly, 2021-12-13:
This issue may be highly relevant to issue #4871 & #4885

rerobika, 2021-12-13:
That's the same issue. You can create any type of assertion here due to the invalid memory access. The fix is about to land soon.

closed 2021-12-13 by rerobika

hope-fly, 2021-12-13:
Okay!