jerryscript-project/jerryscript #4911: how to store binary data in javascript code more efficiency

lee88688, 2021-12-22:
I would like to store some binary data, eg. images or fonts etc. this binary data will be compiled to byte code and store in rom.
currently I tend to use string or Uint8Array, but I don't know which would be more efficiency when runing the compiled to byte code.

dbatyai, 2021-12-22:
It really depends, it would be good to have a bit more information.
How do you intend to put your data into the string or typedarray, and how do you plan to access it later.
Also, in what manner do you mean efficient, performance or memory consumption wise.

lee88688, 2021-12-23:
putting the data into string or typedarray is both ok for me. but I may access it using typedarray later. if I transform the data into typedarray, is it more efficient on memory consumption than string? can performance consumption be distinguished btween typedarray and string?

dbatyai, 2022-01-11:
Since strings are stored internally in cesu-8 encoding, storing the data in a string by mapping each byte to a character would require roughly +50% memory on average. (Values below 128 would be encoded using 1 byte, and values above using 2 bytes).
Since data in this case is stored using variadic encoding, random access into the string would require a linear lookup each time, which would pretty much be horrible performance wise.

Using an Uint8Array requires a constant memory overhead, which consists of the TypedArray object and and the ArrayBuffer header, which can be negligible when storing a lot of data. Access into the buffer will also require only constant overhead. Based on this, TypedArrays should be the preferred choice.

You could use an [external ArrayBuffer](https://github.com/jerryscript-project/jerryscript/blob/master/jerry-core/include/jerryscript-core.h#L700) to store your data in rom, and create a TypedArray with that buffer as backing storage. IMO that would be the most efficient approach both performance and memory wise.

One thing that could cause problems is that (as far as I know) external ArrayBuffers don't have write protection, so if javascript code would try to write to the TypedArray, that could result in a crash.

closed 2022-01-12 by lee88688