jerryscript-project/jerryscript #4917: Use After Free at jerry-core/parser/js/js-lexer.c:3503 in lexer_compare_identifier_to_string

AiDaiP, 2022-01-03:
###### JerryScript revision
a6ab5e9

###### Build platform
Ubuntu 20.04.3 LTS (Linux 5.11.0-43-generic x86_64)

###### Build steps
```sh
./tools/build.py --clean --compile-flag=-fsanitize=address --lto=off --error-message=on --profile=es2015-subset --stack-limit=15 --debug --logging=on --line-info=on
```
###### Test case
```
base64 poc
DQp2YXIgYT0gWyIiLCAiXDAiLCAiXHlmIl0NCkFycmF5LnByb3RvdHlwZVs0XSA9IDEwOw0KZnVu
Y3Rpb24gRWVzdCgpDQp7DQogICAgYS5zb3J0KGZ1bmN0aW9uKCkgew0KICAgQSA9IGZ1bmN0aW9u
KCkgeyB9Ow0KQS5wcm90b3R5cGUueCU9IDQyOw0KdmFyIG8gPSBuZXcgUHJveHkoew0KICAgICAg
ICAiMyI6IHsNCiAgICAgICAgICAgIHdyaXRhYmxlOmZhbHNlLA0KICAgICAgICAgICAgdmFsdUM6
MjANCn0NCiAgICB9LCB7DQogICAgZ2V0UHJvdG90eXBlT2Y6IGZ1bmN0aW9uICh2YWwsIHNpemUs
IGNoKSB7DSAgYS5zb3IKICAgIHZhciBlc3VsdCA9ICIiO3Jlc3VsdCA9IG5ldyBTdHJpUGcgICAg
ICAgICAgICA7Ozs7Ozs7Ozs7OyAgICAgDQogICAgICAgaWYgKGNoID0+IG51bGwpIHsgIGNoID0g
IiAiOw0KfQ0KfSwgew0KICAgIGdldAAAAEBvdHlwZU9oIDwgc2l6ZSAgdWx0Ow0KICAgIH0NCiAg
ICByZXR1cm4gcmVzdWxkOw0KfQ0KfSk7DQogICAgICAgIG8ueCA9IDQzOw0KICAgICAgICB2YXIg
cmVzdWx0ID0gIiI7DQogICAgICAgIGZvciAodmFyIHAgaW4hbykNCiAgICByZXN1bHQgK2EgfCAw
Ow0KICAgIH0pOw0KICAgIFdTY3JpcHQuRWNobyhhKTsNCn0NCg0KVGVzdCgpOw0KDQ==
```

###### Execution steps
```sh
jerry ./poc
```

##### asan log

```
=================================================================
==2953261==ERROR: AddressSanitizer: heap-use-after-free on address 0xf4502508 at pc 0x56731155 bp 0xffa0fc88 sp 0xffa0fc78
READ of size 4 at 0xf4502508 thread T0
    #0 0x56731154 in lexer_compare_identifier_to_string /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-lexer.c:3503
    #1 0x56738c41 in parser_parse_object_literal /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-parser-expr.c:1593
    #2 0x5673c3b8 in parser_parse_unary_expression /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-parser-expr.c:2339
    #3 0x5674559d in parser_parse_expression /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-parser-expr.c:4428
    #4 0x5673e25f in parser_process_unary_expression /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-parser-expr.c:2772
    #5 0x56745605 in parser_parse_expression /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-parser-expr.c:4437
    #6 0x5674e9a0 in parser_parse_var_statement /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-parser-statm.c:527
    #7 0x5675bb93 in parser_parse_statements /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-parser-statm.c:2925
    #8 0x566809a2 in parser_parse_function /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-parser.c:2792
    #9 0x5672d0d9 in lexer_construct_function_object /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-lexer.c:2820
    #10 0x567398f9 in parser_parse_function_expression /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-parser-expr.c:1828
    #11 0x5673c0df in parser_parse_unary_expression /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-parser-expr.c:2313
    #12 0x5674559d in parser_parse_expression /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-parser-expr.c:4428
    #13 0x5673e25f in parser_process_unary_expression /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-parser-expr.c:2772
    #14 0x56745605 in parser_parse_expression /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-parser-expr.c:4437
    #15 0x56745334 in parser_parse_expression_statement /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-parser-expr.c:4399
    #16 0x5675cf43 in parser_parse_statements /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-parser-statm.c:3230
    #17 0x566809a2 in parser_parse_function /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-parser.c:2792
    #18 0x5674fcbb in parser_parse_function_statement /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-parser-statm.c:799
    #19 0x5675bc06 in parser_parse_statements /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-parser-statm.c:2957
    #20 0x5667d2ac in parser_parse_source /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-parser.c:2340
    #21 0x5668356d in parser_parse_script /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/parser/js/js-parser.c:3413
    #22 0x565bd97d in jerry_parse_common /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/api/jerryscript.c:398
    #23 0x565bdce1 in jerry_parse /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/api/jerryscript.c:466
    #24 0x565b6734 in main /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-main/main-jerry.c:161
    #25 0xf7618ee4 in __libc_start_main (/lib/i386-linux-gnu/libc.so.6+0x1eee4)
    #26 0x565b5a94 in _start (/home/aidai/fuzzing/jerryscript/jerryscript-test/build/bin/jerry+0x1ea94)

0xf4502508 is located 8 bytes inside of 124-byte region [0xf4502500,0xf450257c)
freed by thread T0 here:
    #0 0xf7a20814 in __interceptor_free (/lib32/libasan.so.5+0x113814)
    #1 0x56669e18 in jmem_heap_free_block_internal /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/jmem/jmem-heap.c:477
    #2 0x56669f57 in jmem_heap_free_block /home/aidai/fuzzing/jerryscript/jerryscript-test/jerry-core/jmem/jmem-heap.c:691
[... 71 lines cut ...]
  0x3e8a04d0: fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd fd
  0x3e8a04e0: fa fa fa fa fa fa fa fa 00 00 00 00 00 00 00 00
  0x3e8a04f0: 00 00 00 00 00 00 00 04 fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==2953261==ABORTING
```

hope-fly, 2022-01-04:
Another form of PoC

<details>
<summary>poc.js</summary>
<pre><code>

```javascript
function JSEtest() {
  for (let v8 = 0; v8 < 127; v8++) {
      for (let v15 = 2; v15 < 100; v15 = v15 + 10) {
          try {
              let v17 = String;
              const v18 = v17.fromCharCode(10*4, v8, v15, v15, v8);
              const v19 = eval(v18);
          } catch (v20) {
          }
      }
  }
}
JSEtest();
```
</code></pre>
</details>

rerobika, 2022-01-04:
The main issue is similar to #4916. The problem is going to be resolved via #4942.

closed 2022-01-04 by rerobika