jerryscript-project/jerryscript #4932: Assertion 'page_p != NULL' failed at jerryscript/jerry-core/parser/js/js-parser-mem.c(parser_list_get):274.

FlydragonTy, 2022-01-04:
###### JerryScript revision

Commit: [a6ab5e9](https://github.com/jerryscript-project/jerryscript/commit/a6ab5e9abed70cdedf9f4e9c1dc379eb762ebf64)

Version: v3.0.0

###### Build platform

Ubuntu 18.04.5 LTS (Linux 4.19.128-microsoft-standard x86_64)

Ubuntu 18.04.5 LTS (Linux 5.4.0-44-generic x86_64)

###### Build steps

```bash
python ./tools/build.py --clean --debug --compile-flag=-fsanitize=address --compile-flag=-m32 --compile-flag=-g --strip=off --lto=off --logging=on --line-info=on --error-message=on --system-allocator=on --stack-limit=20
```

###### Test case
<details>
<summary>poc.js</summary>
<pre><code>

```javascript
class JSEtest {
  async *m() { return 42; } #m() { return 'test262'; }; //  method() {
    return this.#m();
  }
}

var c = new JSEtest();

assert.sameValue(Object.hasOwnProperty.call(c, "m"), false);
assert.sameValue(c.m, JSEtest.prototype.m);

verifyProperty(JSEtest.prototype, "m", {
  enumerable: false,
  configurable: true,
  writable: true,
}, { restore: true });

c.m().next().then(function (v) {
  assert.sameValue(v.value, 42);
  assert.sameValue(v.done, true);

  function assertions() {
    // Cover $DONE handler for async cases.
    function $DONE(error) {
      if (error) {
        throw new Test262Error('Test262:AsyncTestFailure')
      }
    }
    assert.sameValue(c.method(), 'test262');
  }

  return Promise.resolve(assertions());
}, $DONE).then($DONE, $DONE);

```
</code></pre>
</details>
​	

###### Execution steps & Output

```bash
$ ./jerryscript/build/bin/jerry poc.js

ICE: Assertion 'page_p != NULL' failed at jerryscript/jerry-core/parser/js/js-parser-mem.c(parser_list_get):274.
Error: ERR_FAILED_INTERNAL_ASSERTION
[1]    7036 abort      jerry poc.js
```

rerobika, 2022-01-04:
Looks quite similar to #4930 and #4931

rerobika, 2022-01-04:
Duplicate of #4930.

closed 2022-01-04 by rerobika