jerryscript-project/jerryscript #4963: Proposal: Retire the legacy ES5.1 build profile

ossy-szeged, 2022-01-11:
JerryScript is a full ES5.1 compliant JavaScript engine since its first release (JerryScript 1.0 in 2016). JavaScript language is continuously evolving: the first popular version (5.1) was released in 2011 and new releases have been rolled out every year since 2015. JavaScript engines are evolving too, JerryScript finished implementing ES2015 features in 2020 and then the project has been retargeted (#3737) to ES.Next (always to the latest ECMAScript specification). Now 83.6% of the Official Conformance Test Suite (Test262) pass - which contains tests for Stage 4 and Stage 3 features too (features will be released in the next or any future specification).

Since there are many incompatible changes between ECMAScript versions, furthermore the ES5.1 specification is more than 10 years old, we would like to drop its support. A pull request is already prepared for this cleanup: #4915. If there are no reasonable objections within two weeks, we would like to drop it.

LaszloLango, 2022-01-28:
I have no objections, but I'd like to suggest that put a tag on the last stable commit which contains the ES5.1 build profile. So if someone needs it in the future then they can find it and rollback to it easily.

rerobika, 2022-01-31:
No es5.1 related development happened since the 2.4 release so it's a suitable tag for this purpose.

ossy-szeged, 2022-02-01:
Already merged in 4924f9fd315e9281a89d0621f278f3a6c1c7efd2

closed 2022-02-01 by ossy-szeged