jerryscript-project/jerryscript #4974: jerry_call with Jerry_Array argument will causing JERRY_FATAL_REF_COUNT_LIMIT randomly

weixiongmei, 2022-02-06:
###### JerryScript revision
#4973

###### Test case
**Error Code**
```c
//periodically calling the following code every 1 second will causing JERRY_FATAL_REF_COUNT_LIMIT randomly
        jerry_value_t jerry_arguments[2];
        jerry_arguments[0] = jerry_number(0);
        jerry_arguments[1] = jerry_array(1);
        jerry_value_t element_0 = jerry_number(0);
        jerry_value_free(jerry_object_set_index(jerry_arguments[1], 0,
            element_0
        ));
        //jerry_value_free(element_0); //Also tried with this line un-commended, but having the same result
        jerry_call(jerry_function, jerry_undefined(), jerry_arguments, 2);
```

**Good Code**
```c
//periodically calling the following code every 1 second will NOT causing JERRY_FATAL_REF_COUNT_LIMIT
        jerry_value_t jerry_arguments[2];
        jerry_arguments[0] = jerry_number(0);
        jerry_arguments[1] = jerry_number(1);
        jerry_call(jerry_function, jerry_undefined(), jerry_arguments, 2);
```
###### Execution steps
Call the above code in the Error Section every 1 second, it will randomly causing JERRY_FATAL_REF_COUNT_LIMIT within an hour.

dbatyai, 2022-02-07:
It would be good to know what the function that you're calling does.
`jerry_call` returns with the value returned by the function, it's possible that needs to be freed as well.

Based on the error message, the most likely cause is that the Array object is leaking at some point, however based on the above code alone it's not possible to tell where exactly.

weixiongmei, 2022-02-08:
> 

Hi, it's just an empty javascript function, and has not return. Here's the code.

```javascript
let test_func = function(arg0, arg1) { };
```

galpeter, 2022-02-08:
Hi,

I've created a simple example application based on the details you provided (with minor modifications), see:
```c
#include "jerryscript.h"

int main (void)
{ 
  jerry_init (JERRY_INIT_EMPTY);

  const jerry_char_t script[] = "function test_func(arg0, arg1) { };";
  jerry_value_t parsed_code = jerry_parse(script, sizeof (script) - 1, NULL);
  if (!jerry_value_is_error(parsed_code))
  { 
    jerry_value_t ret_value = jerry_run(parsed_code);
    jerry_value_free(ret_value);
  }
  jerry_value_free(parsed_code);

  // Get the test function
  jerry_value_t global = jerry_current_realm();
  jerry_value_t jerry_function_demo = jerry_object_get_sz (global, "test_func");
  jerry_value_free (global);

  for (int i = 0; i < 1000000; i++) {
    jerry_value_t jerry_arguments[2];
    jerry_arguments[0] = jerry_number(12321211414.3);
    jerry_arguments[1] = jerry_array(1);
    jerry_value_t element_0 = jerry_number(3.1231231212);
    jerry_value_free( jerry_object_set_index(jerry_arguments[1], 0, element_0) );

    jerry_value_free(element_0); // This is required

    jerry_value_t result = jerry_call(jerry_function_demo, jerry_undefined(), jerry_arguments, 2);
    jerry_value_free(result); // result must be freed

    jerry_value_free(jerry_arguments[0]);
    jerry_value_free(jerry_arguments[1]);
  }

  jerry_value_free(jerry_function_demo);

  jerry_cleanup();

  return 0;
}
```

Using this, I was unable to reproduce the ref count limit error. Still a few important key points:
* `jerry_call`'s result must be freed
* values created via `jerry_number` must be also freed (there are some cases where this could be skipped if the number is a very small number, but that is an internal thing and can change, so don't build on skipping free)

How do you load the`test_func` js function's value into a `jerry_value_t`? (which API calls are you using?)

weixiongmei, 2022-02-08:
Hi @galpeter, thank you very much, seems like my problem is half way solved, i didn't freed the returned value of jerry_call, and didn't freed all the items of an array, but still having problem if the array contains jerry_string like the code below, would you mind to test it with a jerry_string item in the array? Thanks

```c
char* test_string = "jerry test string";
jerry_value_t element_1 = jerry_string((jerry_char_t*)test_string, strlen(test_string), JERRY_ENCODING_UTF8);
jerry_value_free( jerry_object_set_index(jerry_arguments[1], 1, element_1));
jerry_value_free(element_1);
```

galpeter, 2022-02-09:
I've added the code part you mentioned to the example above (just before the jerry_call) and it works without any errors. I think there the problem lies somewhere else.

closed 2022-02-11 by weixiongmei