jerryscript-project/jerryscript #4975: temporal.now require ` nanosecond accuracy`

lygstate, 2022-02-09:
According to
https://tc39.es/proposal-temporal/docs/now.html#instant,

The Temporal.Now.instant() are improved the  accuracy over Date.now from
millisecond to nanosecond, that's need tthe `jerryscript`'s port API to support nanosecond precision

dbatyai, 2022-02-09:
According to the proposal:

> Returns: a Temporal.Instant object representing the current system time.

I'm going to refer to [my previous comment](https://github.com/jerryscript-project/jerryscript/pull/4907#discussion_r784985865) on this topic.

Is there actually a platform in existence that supports true nanosecond clock resolution for the system clock? So far I have yet to see any, but if I'm wrong then please correct me.

As such, even if the proposed Temporal object is able to store a time value in nanosecond precision does not mean that underlying hardware has sufficiently high precision timers, or that operating systems provide a way to query timers in that precision.

I think the way the proposal should be interpreted is the following: Take the current system time using the highest precision clock available, and scale it to nanoseconds. For this purpose the current port function is already more than sufficient.

lygstate, 2022-02-09:
> According to the proposal:
>
>
> I'm going to refer to [my previous comment](https://github.com/jerryscript-project/jerryscript/pull/4907#discussion_r784985865) on this topic.
>
> Is there actually a [...]

Hi, on windows, the system clock is 100ns, on most posix platform, it's nanosecond, please refer to
[clock_gettime](https://linux.die.net/man/3/clock_gettime)

> As such, even if the proposed Temporal object is able to store a time value in nanosecond precision does not mean that underlying hardware has sufficiently high precision timers, or that operating [...]

dbatyai, 2022-02-09:
Well, in my experience the `tv_nsec` member never actually contains nanosecond precision values, it's always a multiple of 1000. I'm also rather skeptical about the 100ns resolution in Windows, I would say it's also pretty likely that it's not fully utilized.

However you are correct that at least there are possibilities to query in (close to) nanosecond precision. I guess it would probably make sense to update the port function to resemble `clock_gettime`, just in case, even if we can't fully utilize it yet.