jerryscript-project/jerryscript #4983: on unix, realpath are can failed when the path not exist

lygstate, 2022-02-10:
```
jerry_char_t *
jerry_port_path_normalize (const jerry_char_t *path_p, /**< input path */
                           jerry_size_t path_size) /**< size of the path */
{
  (void) path_size;

  return (jerry_char_t *) realpath ((char *) path_p, NULL);
} /* jerry_port_path_normalize */
```
Currently the implementation of jerry_port_path_normalize  are called realpath directly, than
may return NULL when path_p is not NULL, but that's not expected.

LaszloLango, 2024-11-22:
Do you have any specific solution for this? Would you like open a PR for this?

lygstate, 2024-11-22:
> Do you have any specific solution for this? Would you like open a PR for this?

I fixed it locally by using https://github.com/lygstate/cpathjoin

that is do not use operating system provided fullppath realptah,
so we only need to export a single api
```
/**
 * Get the current working directory
 *
 * Note:
 *      Refer to https://pubs.opengroup.org/onlinepubs/9699919799/functions/getcwd.html
 *
 * @param out_buf_p Pointer to the output buffer where the normalized path should be written.
 * @param out_buf_size Size of the output buffer.
 *
 * @return The current working directory path
 */
char *jerry_port_get_cwd (char *out_buf_p, size_t out_buf_size);
```
is enough,
indeed, the module handling needs path.join
and nodejs have https://nodejs.org/api/path.html#windows-vs-posix
so to properly implement we needs nodejs alike path join

lygstate, 2024-12-11:
> Do you have any specific solution for this? Would you like open a PR for this?

@LaszloLango PR prepared, better merged as Release 3.0, as 
It's update the port api

LaszloLango, 2024-12-12:
We need to wrap up things for the release, we won't be able to review and merge #5198 in time, so it'll be scheduled to the next one.