jerryscript-project/jerryscript #4994: A question about Array.prototype.sort

YiWen-y, 2022-04-09:
###### Version

Version: 3.0.0 (fea10bb7)

###### Test case

```js
var foo = function (parts) {
    parts.sort(function (a, b) {
        return a.avalue - b.value;//NaN
    });
    parts.forEach(function (item) {
        print(item.value)
    })
};

var Parameter0 = [
    { name: 'Edward', value: 21 },
    { name: 'Tom', value: 37 },
    { name: 'Jerry', value: -12 }
];
foo(Parameter0);
```

###### Execution steps 

    /.jsvu/jerry Testcase.js

###### Output

```
-12 37 21
```

###### Expected behavior

```
21 37 -12
```

###### Description

When executing this test case, I mistakenly wrote `a.value` to `a.avalue` in line 3, which caused all the return values of `parts.sort` to become `NaN`. At this point, none of the other engines changed the ordering of the original array, but `jerryscript` reversed the array.

AidPaike, 2022-10-08:
https://github.com/jerryscript-project/jerryscript/issues/5025#issue-1386947482

AidPaike, 2022-10-08:
The author replied to us  https://github.com/jerryscript-project/jerryscript/issues/5025#issue-1386947482

closed 2022-10-09 by YiWen-y