jerryscript-project/jerryscript #4996: How I can reload script?

anzenkovich, 2022-04-11:
Hey!

I am using jerry script as a scripting backend in my game engine. And it works great :)

But I can't figure out how to implement script reloading. When you develop something, you need to restart and run scripts over and over again. How can i do this?

In other scripting languages, I just loaded the script again and all objects were overwritten. But Jerry says, “OK, this property already exists. Stop it." This happens when I try to reload the classes.

I think I can fix the source code and change this behaviour, but is this the correct solution? How can I load a script with some class and then reload that script with the same class?

rerobika, 2022-04-14:
Hi!

Instead of reloading the engine you can simply change the current realm. Please check https://github.com/jerryscript-project/jerryscript/blob/master/docs/02.API-REFERENCE.md#jerry_set_realm.

anzenkovich, 2022-04-17:
Yes, thank you, but I need something else :) I need to reload just one file of several. For example, I have several files, where declared some classes:

file_a.js:
```
class a
{
  constructor() { print('hello a'); }
  update(dt) { ... do some things ... }
}
```

file_b.js:
```
class mySomeClass
{
  constructor() { this.interestingThing = 'abcda'; }
}
class b
{
  constructor() { this.x = new mySomeClass(); }
  update(dt) { print('dt is ' + dt + ', thing is ' + this.x); }
}
```

somewhere I create instances of this classes. Then I want to reload file_b.js, cause I changed it. So I can edit some scripts without restarting the engine.

I found some workaround, how I can use it, but it is not so cool :) I can directly assign classes to global variables with same name:
a = class a { ... }