jerryscript-project/jerryscript #5015: import symbol conflicts with global variable name

dananderson, 2022-07-30:
###### JerryScript revision
2.4.0

###### Build platform
all

###### Build steps
n/a

###### Build log
n/a

###### Test case
Assume there is a global variable named 'process' in the current realm. Assume the process package exports default and a symbol named process as well.

When using a default import:

    import process from 'process'

When using a named import: 

    import { process } from 'process'

Both statements fail.

###### Execution platform
all

###### Execution steps
n/a

###### Output
Error message: Imported binding shadows local variable

###### Backtrace
n/a

###### Expected behavior
No parse or runtime Error.

zherczeg, 2022-09-08:
It looks like this is a "feature" in JerryScript. Probabyl something was misunderstood. Attempting to change this in #5023

zherczeg, 2022-09-10:
Patch landed. Thank you for the report.

closed 2022-09-10 by zherczeg