jerryscript-project/jerryscript #5018: Import and Export statement

DiamondToken, 2022-07-31:
how to use import statement without syntax error in v2.4.0?

zherczeg, 2022-08-01:
Did you pass -m to jerryscript? From api, you need to pass a flag to jerry_parse.

DiamondToken, 2022-08-01:
i made it import and export, but with different parse options i can get either import, export or scope.
if i parse with JERRY_PARSE_MODULE i can import but my scope is empty
if i parse with JERRY_PARSE_NO_OPTS i can see some variables in the scope via globalThis

in both cases i can't see variables and functions from c code like this

`   
    interpret = jerryx_get_property_str(global_obj_val, "interpret");

    jerry_release_value(global_obj_val);

    if (jerry_value_is_function(interpret)) 
`

zherczeg, 2022-08-01:
Imported variables goes to a local scope, they are not part of the global object. The `jerry_module_namespace` can be used to get the export namespace. If you also export the imported symbols, they are available there. A module can also export a helper function with eval to get the local variables (or do anything else): `export function access(s) { return eval(s) }`

dananderson, 2022-08-02:
Imports are different from normal requires. The process is as follows:

jerry_parse() with JERRY_PARSE_MODULE option returns a module object. The module object is in the unlinked state. Next, you need to call jerry_module_link() on the module. That moves it to the linked state (and resolves any imports of that module). Finally, call jerry_module_evaluate() to run the module.

If you want the exports, such as your interpret method, use jerry_module_namespace() to get the exports from C.

The jerry implementation follows the ECMA spec with respect to modules. But, making a usable import system takes some work.

DiamondToken, 2022-08-02:
in v2.4.0 there is no such thing like jerry_module_link.
I want to get function which lives in JS from C.

dananderson, 2022-08-02:
Oh, I did not realize 2.4.0 was so outdated. I don't see a public api in 2.4.0 for accessing an es module's exports.

DiamondToken, 2022-08-03:
anyway, many thanks to you! @dananderson @zherczeg

closed 2022-08-03 by DiamondToken