jerryscript-project/jerryscript #5020: Build failed with clang 3.8: error: missing field 'id' initializer [-Werror,-Wmissing-field-initializers]

zalsaeed, 2022-08-18:
## Brief Description & Steps

Failed to build on Debian with Clang 3.8. In the same environment the build succeeds with cc/gcc, but not with the version of clang used . This is the error I'm getting `error: missing field 'id' initializer [-Werror,-Wmissing-field-initializers]` on `ecma-builtin-typedarray-prototype.c:1922:37`. Below is the full detail of the error and how to regenerate it.

#### JerryScript revision
`368641043fc9580400a179669f9b45782db90538`

#### Build platform
* Environment: `Debian GNU/Linux 9 (stretch) - (Linux 5.10.104-linuxkit x86_64)`
* Clang: `clang version 3.8.1-24 (tags/RELEASE_381/final)`

#### Build steps
```sh
CC=clang python tools/build.py --debug --verbose --clean 
```

#### Partial Build Stdout
```bash
[ 60%] Building C object jerry-core/CMakeFiles/jerry-core.dir/ecma/builtin-objects/typedarray/ecma-builtin-uint16array-prototype.c.o
cd /home/jerryscript/build/jerry-core && /usr/bin/clang  -DJERRY_CPOINTER_32_BIT=0 -DJERRY_DEBUGGER=0 -DJERRY_ERROR_MESSAGES=0 -DJERRY_EXTERNAL_CONTEXT=0 -DJERRY_FUNCTION_TO_STRING=0 -DJERRY_GC_LIMIT="(0)" -DJERRY_GC_MARK_LIMIT="(8)" -DJERRY_GLOBAL_HEAP_SIZE="(512)" -DJERRY_LINE_INFO=0 -DJERRY_LOGGING=0 -DJERRY_MEM_GC_BEFORE_EACH_ALLOC=0 -DJERRY_MEM_STATS=0 -DJERRY_PARSER=1 -DJERRY_PARSER_DUMP_BYTE_CODE=0 -DJERRY_PROMISE_CALLBACK=0 -DJERRY_REGEXP_DUMP_BYTE_CODE=0 -DJERRY_REGEXP_STRICT_MODE=0 -DJERRY_SNAPSHOT_EXEC=0 -DJERRY_SNAPSHOT_SAVE=0 -DJERRY_STACK_LIMIT="(0)" -DJERRY_SYSTEM_ALLOCATOR=0 -DJERRY_VALGRIND=0 -DJERRY_VM_HALT=0 -DJERRY_VM_THROW=0 -I/home/jerryscript/jerry-core/include -I/home/jerryscript/jerry-core -I/home/jerryscript/jerry-core/api -I/home/jerryscript/jerry-core/debugger -I/home/jerryscript/jerry-core/ecma/base -I/home/jerryscript/jerry-core/ecma/builtin-objects -I/home/jerryscript/jerry-core/ecma/builtin-objects/typedarray -I/home/jerryscript/jerry-core/ecma/operations -I/home/jerryscript/jerry-core/jcontext -I/home/jerryscript/jerry-core/jmem -I/home/jerryscript/jerry-core/jrt -I/home/jerryscript/jerry-core/lit -I/home/jerryscript/jerry-core/parser/js -I/home/jerryscript/jerry-core/parser/regexp -I/home/jerryscript/jerry-core/vm  -flto -std=c99 -pedantic -Wall -Werror=all -Wextra -Werror=extra -Wformat-nonliteral -Werror=format-nonliteral -Winit-self -Werror=init-self -Wsign-conversion -Werror=sign-conversion -Wformat-security -Werror=format-security -Wmissing-declarations -Werror=missing-declarations -Wshadow -Werror=shadow -Wstrict-prototypes -Werror=strict-prototypes -Wundef -Werror=undef -Wold-style-definition -Werror=old-style-definition -Wconversion -Werror=conversion -Wno-stack-protector -Wno-attributes -Werror -Wno-nested-anon-types -Wno-static-in-inline -g   -o CMakeFiles/jerry-core.dir/ecma/builtin-objects/typedarray/ecma-builtin-uint16array-prototype.c.o   -c /home/jerryscript/jerry-core/ecma/builtin-objects/typedarray/ecma-builtin-uint16array-prototype.c
/home/jerryscript/jerry-core/ecma/builtin-objects/typedarray/ecma-builtin-typedarray-prototype.c:1922:37: error: missing field 'id' initializer [-Werror,-Wmissing-field-initializers]
  ecma_typedarray_info_t info = { 0 };
                                    ^
1 error generated.
[ 61%] Building C object jerry-core/CMakeFiles/jerry-core.dir/ecma/builtin-objects/typedarray/ecma-builtin-uint32array-prototype.c.o
jerry-core/CMakeFiles/jerry-core.dir/build.make:3065: recipe for target 'jerry-core/CMakeFiles/jerry-core.dir/ecma/builtin-objects/typedarray/ecma-builtin-typedarray-prototype.c.o' failed
make[2]: *** [jerry-core/CMakeFiles/jerry-core.dir/ecma/builtin-objects/typedarray/ecma-builtin-typedarray-prototype.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
cd /home/jerryscript/build/jerry-core && /usr/bin/clang  -DJERRY_CPOINTER_32_BIT=0 -DJERRY_DEBUGGER=0 -DJERRY_ERROR_MESSAGES=0 -DJERRY_EXTERNAL_CONTEXT=0 -DJERRY_FUNCTION_TO_STRING=0 -DJERRY_GC_LIMIT="(0)" -DJERRY_GC_MARK_LIMIT="(8)" -DJERRY_GLOBAL_HEAP_SIZE="(512)" -DJERRY_LINE_INFO=0 -DJERRY_LOGGING=0 -DJERRY_MEM_GC_BEFORE_EACH_ALLOC=0 -DJERRY_MEM_STATS=0 -DJERRY_PARSER=1 -DJERRY_PARSER_DUMP_BYTE_CODE=0 -DJERRY_PROMISE_CALLBACK=0 -DJERRY_REGEXP_DUMP_BYTE_CODE=0 -DJERRY_REGEXP_STRICT_MODE=0 -DJERRY_SNAPSHOT_EXEC=0 -DJERRY_SNAPSHOT_SAVE=0 -DJERRY_STACK_LIMIT="(0)" -DJERRY_SYSTEM_ALLOCATOR=0 -DJERRY_VALGRIND=0 -DJERRY_VM_HALT=0 -DJERRY_VM_THROW=0 -I/home/jerryscript/jerry-core/include -I/home/jerryscript/jerry-core -I/home/jerryscript/jerry-core/api -I/home/jerryscript/jerry-core/debugger -I/home/jerryscript/jerry-core/ecma/base -I/home/jerryscript/jerry-core/ecma/builtin-objects -I/home/jerryscript/jerry-core/ecma/builtin-objects/typedarray -I/home/jerryscript/jerry-core/ecma/operations -I/home/jerryscript/jerry-core/jcontext -I/home/jerryscript/jerry-core/jmem -I/home/jerryscript/jerry-core/jrt -I/home/jerryscript/jerry-core/lit -I/home/jerryscript/jerry-core/parser/js -I/home/jerryscript/jerry-core/parser/regexp -I/home/jerryscript/jerry-core/vm  -flto -std=c99 -pedantic -Wall -Werror=all -Wextra -Werror=extra -Wformat-nonliteral -Werror=format-nonliteral -Winit-self -Werror=init-self -Wsign-conversion -Werror=sign-conversion -Wformat-security -Werror=format-security -Wmissing-declarations -Werror=missing-declarations -Wshadow -Werror=shadow -Wstrict-prototypes -Werror=strict-prototypes -Wundef -Werror=undef -Wold-style-definition -Werror=old-style-definition -Wconversion -Werror=conversion -Wno-stack-protector -Wno-attributes -Werror -Wno-nested-anon-types -Wno-static-in-inline -g   -o CMakeFiles/jerry-core.dir/ecma/builtin-objects/typedarray/ecma-builtin-uint32array-prototype.c.o   -c /home/jerryscript/jerry-core/ecma/builtin-objects/typedarray/ecma-builtin-uint32array-prototype.c
```

#### Dockerfile to Rebuild the same Environment

###### Dockerfile
```dockerfile
FROM debian:9.8

RUN apt-get update
RUN apt-get install -y --no-install-recommends \
	build-essential \
	make \
	cmake \
	git \
	curl \
	vim \
	wget \ 
	ca-certificates \
	python \ 
	clang

WORKDIR /home
RUN git clone https://github.com/jerryscript-project/jerryscript.git
```

###### Build Docker Image

```bash
docker build -t jerryscript:latest .
```

###### Run Docker Container

```bash
docker run --name js -it jerryscript /bin/bash
```

## Complete Log

###### CMakeOutput.log

[... 218 lines cut ...]
/usr/bin/make -f CMakeFiles/cmTC_199e6.dir/build.make CMakeFiles/cmTC_199e6.dir/build
make[1]: Entering directory '/home/jerryscript/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_199e6.dir/src.c.o
/usr/bin/clang  -D_BSD_SOURCE -D_DEFAULT_SOURCE  -flto -std=c99 -pedantic -Wall -Werror=all -Wextra -Werror=extra -Wformat-nonliteral -Werror=format-nonliteral -Winit-self -Werror=init-self -Wsign-conversion -Werror=sign-conversion -Wformat-security -Werror=format-security -Wmissing-declarations -Werror=missing-declarations -Wshadow -Werror=shadow -Wstrict-prototypes -Werror=strict-prototypes -Wundef -Werror=undef -Wold-style-definition -Werror=old-style-definition -Wconversion -Werror=conversion -Wno-stack-protector -Wno-attributes -Werror -Wno-nested-anon-types -Wno-static-in-inline -DHAVE_TM_GMTOFF -Wno-error=strict-prototypes -Wno-error=old-style-definition -Wno-error=unused-value   -o CMakeFiles/cmTC_199e6.dir/src.c.o   -c /home/jerryscript/build/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_199e6
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_199e6.dir/link.txt --verbose=1
/usr/bin/clang   -flto -std=c99 -pedantic -Wall -Werror=all -Wextra -Werror=extra -Wformat-nonliteral -Werror=format-nonliteral -Winit-self -Werror=init-self -Wsign-conversion -Werror=sign-conversion -Wformat-security -Werror=format-security -Wmissing-declarations -Werror=missing-declarations -Wshadow -Werror=shadow -Wstrict-prototypes -Werror=strict-prototypes -Wundef -Werror=undef -Wold-style-definition -Werror=old-style-definition -Wconversion -Werror=conversion -Wno-stack-protector -Wno-attributes -Werror -Wno-nested-anon-types -Wno-static-in-inline -DHAVE_TM_GMTOFF -Wno-error=strict-prototypes -Wno-error=old-style-definition -Wno-error=unused-value    CMakeFiles/cmTC_199e6.dir/src.c.o  -o cmTC_199e6 -rdynamic 
/usr/bin/ld: /usr/lib/llvm-3.8/bin/../lib/LLVMgold.so: error loading plugin: /usr/lib/llvm-3.8/bin/../lib/LLVMgold.so: cannot open shared object file: No such file or directory
clang: error: linker command failed with exit code 1 (use -v to see invocation)
CMakeFiles/cmTC_199e6.dir/build.make:97: recipe for target 'cmTC_199e6' failed
make[1]: *** [cmTC_199e6] Error 1
make[1]: Leaving directory '/home/jerryscript/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_199e6/fast' failed
make: *** [cmTC_199e6/fast] Error 2

Source file was:

#include <time.h>

int main()
{
   (void)sizeof(((struct tm *)0)->tm_gmtoff);
   return 0;
}
```

zherczeg, 2022-08-18:
It looks like you have found a compiler bug. In C, partial initialization of struct is possible, and the uninitialized members are zeroed. So creating a full zero data with `= { 0 }` is possible regardless of struct.

zalsaeed, 2022-08-18:
I see that others were able to build JerryScript with newer versions of clang (thus it seems fixed for newer versions). Clang 3.8 is not going to change. Thus there are two possible fixes here. 

## Change all the initialization to be explicit with the number of elements in all structs
Luckily there was only two initialization that required changes.

1. https://github.com/jerryscript-project/jerryscript/blob/368641043fc9580400a179669f9b45782db90538/jerry-core/ecma/builtin-objects/typedarray/ecma-builtin-typedarray-prototype.c#L1922
The line should be changed to  `ecma_typedarray_info_t info = { 0,0,0,0,0,0 };`
2. https://github.com/jerryscript-project/jerryscript/blob/368641043fc9580400a179669f9b45782db90538/jerry-core/parser/js/js-parser.c#L2925
The line should be changed to `scanner_range_t range = { 0,{0,0,0} };`

I didn't put these in a pull request as it would require running tests on several other environments (which I don't have) to ensure it is still compatible. And I'm not actively working on JerryScript. However, feel free to apply them if you find this acceptable. 

## Suppress (disable) the warnings/errors with Cmake
Unfortunately, I cannot find the proper way to apply such flags with the python build script. Any help here is highly appreciated.

-------
My problem is solved for now. Please feel free to close this issue if the changes are not going to be applied.