jerryscript-project/jerryscript #5022: Builds

89z, 2022-09-02:
Are any builds available? I am looking for Windows

akosthekiss, 2022-09-02:
@89z The project can be built for Windows (actually, every commit is built on AppVeyor CI, see https://ci.appveyor.com/project/jerryscript-project/jerryscript). However, the CI build does not produce artifacts.

As the build does not need anything but a C compiler and CMake, it should be quite easy to build it for yourself. The AppVeyor CI config file (https://github.com/jerryscript-project/jerryscript/blob/master/appveyor.yml) is a good starting point how to build the project on/for Windows.

89z, 2022-09-02:
I am trying to build without Visual Studio, perhaps using LLVM or GNU

akosthekiss, 2022-09-02:
Then `tools/build.py` _may_ work for you (it requires python in addition to cmake and the compiler). Please, note that this configuration is not tested. It is not impossible that it will work out of the box, but it can just as easily happen that you bump into some platform-specific issue that needs tweaks or workarounds. Anyway, if it works for you, we'd love to hear about it.