jerryscript-project/jerryscript #5030: Can't make JerryScript Work with Zephyr

ig-66, 2022-10-26:
I'm simply not able to make JerryScript work with Zephyr. I have tried different processes with different outcomes:

### 1> 
Tried running the command below (just like this [tutorial](https://github.com/jerryscript-project/jerryscript/tree/master/targets/os/zephyr)) in the `targets\os\zephyr`, but changing the toolchain in the CMakeLists:
```
west build -p auto -b waveshare_nrf5240_eval_kit
```
Changes to the CMakeLists:
from this:
`--toolchain=${JERRY_BASE}/cmake/toolchain_mcu_stm32f4.cmake)`
to this:
`--toolchain=${ZEPHYR_BASE}/boards/arm/waveshare_nrf52840_eval_kit/board.cmake)`

Result:
```
Build files have been written to: C:/company/projects/tst/west-workspace/jerryscript/targets/os/zephyr/build
ninja: error: '../../../../build/lib/libjerry-core.a', needed by 'zephyr/zephyr_prebuilt.elf', missing and no known rule to make it
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'C:\company\projects\tst\west-workspace\jerryscript\targets\os\zephyr\build
```

### 2> 
So, tried running the command to generate the libjerry files from the `jerryscript` folder:
`python tools\build.py`, but this gave me this error:
```
[...]
running 'nmake' '-?' failed with the system cannot find the file specified
[...]
```
So I set up the python build to use the ninja (because Zephyr uses ninja, and it is already configured in my machine) using CMake GUI, also set the following with CMake GUI:
```
CMAKE_C_COMPILER <= C:/GNU_ARM_EMBEDDED/bin/arm-none-eabi-gcc.exe
CMAKE_C_COMPILER_WORKS <= TRUE
```
Also set the [configs](https://github.com/jerryscript-project/jerryscript/blob/6fe763f1912c3ef82ec2ca31c40f186dde9c5a57/targets/os/zephyr/CMakeLists.txt#L23) in the `jerryscript\targets\os\zephyr\CMakeLists.txt`  in CMake GUI

And then ran the command `python tools\build.py`, the build is succeeded. After that tried building `jerryscript\targets\os\zephyr` with west.

Result:
```
[...]
fatal error: sys/printk.h: No such file or directory
   19 | #include <sys/printk.h>
[...]
```

### 3>
Then tried running west command in the jerryscript directory (after deleting the contents in the build folder, of course):
```
west build -p auto -b waveshare_nrf52840_eval_kit
```
The build was succeeded, but got the following warning:
```
CMake Warning:
  Manually-specified variables were not used by the project:

    BOARD
    WEST_PYTHON

-- Build files have been written to: C:/moduhub/projects/jerryscript_test/jerrytest/jerryscript/build
-- west build: building application
[3/29] Generating ../amalgam/jerryscript.c, ../amalgam/jerryscript.h
WARNING:root:[2] Include not found (C:\company\projects\jerryscript_test\jerrytest\jerryscript\jerry-core\jmem\jmem-allocator-internal.h:32): "memcheck.h"
[29/29] Linking C static library lib\libjerry-ext.a
```
Then returned to `targets\os\zephyr`, removed the build folder and ran:
```
west build -p auto -b waveshare_nrf52840_eval_kit
```
Got:
```
[...]
fatal error: sys/printk.h: No such file or directory
   19 | #include <sys/printk.h>
[...]
```
So I removed the `#include <sys/printk.h>` header, as it is not required for `printk` anymore, and re-ran the west command.

Result:
```
[...]
fatal error: zephyr.h: No such file or directory
   19 | #include <zephyr.h>
[...]
```

### 4>
Update Zephyr to 3.1.0, setup the the build in the jerryscript folder with CMake GUI, the same way as described above and ran west in `targets\os\zephyr`

Result:
```
c:/gnu_arm_embedded/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: error: zephyr\zephyr_prebuilt.elf uses VFP register arguments, ../../../../build/lib/libjerry-core.a(jerryscript.c.obj) does not
```
But that was fixed by setting float ABI to use soft float calling conventions using `menuconfig`/`guiconfig`:
[image]

After that I re-ran the west build command, the build was succeeded, so I flashed to the board and openned the Arduino's Serial Monitor, the program simply stops, note how the "js>" is not printed:
[image]
Tried sending `print('Test')` through the serial monitor, but nothing happens.
When debbug with Ozone, the problem is in the `jerry_init()` function in the `jerry-main.c`, a MemManage exception heppens:
[image]
Maybe the warning in try number 3 has something to do with it:
```
[...]
WARNING:root:[2] Include not found (C:\company\projects\jerryscript_test\jerrytest\jerryscript\jerry-core\jmem\jmem-allocator-internal.h:32): "memcheck.h"
[...]
``` 

### Comments
I haven't setup Zephyr according to the JerryScript tutorial, because Zephyr is already setup and running.

Platform: Windows 10
Zephyr version: 2.7.0 and 3.1.0
JerryScript Version: 3.0.0

ig-66, 2022-11-09:
Are you guys alive overthere?

akosthekiss, 2022-11-09:
Well, yes, thanks for asking. A gentle ping would have sufficed.

zherczeg, 2022-11-10:
It seems system headers are missing, maybe something is not installed. We don't know your platform, but somebody at zephyr community might know more.

ig-66, 2022-11-10:
> Well, yes, thanks for asking. A gentle ping would have sufficed.

@akosthekiss Sorry, it sounded funny on my head

ig-66, 2022-11-10:
I asked overthere yesterday, but as it isn't considered a Zephyr issue, that got moved to the discussion, still nothing though it is recent, so that is expected. BTW I'm working on making Jerry a Zephyr module, lets see how that turn out

zherczeg, 2022-11-11:
Unfortunately I cannot really comment on the "top of stack" issue. The memcheck.h is needed by valgrind, did you enabled its support?

zherczeg, 2022-11-11:
A stack issue just came to my mind. I had a multi threaded experiment before, where I accidentally placed the whole jerry heap on thread stack, and it crashed instantly. This is an unlikely case, but maybe it helps.

ig-66, 2022-11-14:
> Unfortunately I cannot really comment on the "top of stack" issue. The memcheck.h is needed by valgrind, did you enabled its support (passing `JERRY_VALGRIND` to the compiler)?

Actually I haven't done that. But as of now, the Jerry module for zephyr is working to some extent, but I have some features to fix. Either way, thanks for the help!

ig-66, 2023-01-02:
Update: 
Here is a repository of a modified version of [JerryScript to be built for Zephyr](https://github.com/moduhub/zephyr_jerryscript). * It might suffer modifications, but it is working great so far *
And here is a test/sample repo on how to use [JerryZ](https://github.com/ig-66/jerry_z_tests)

closed 2023-01-02 by ig-66

rtakacs, 2023-01-03:
Hi there, I've tried out the current Zephyr version (2.7) with latest JerryScript (that is for stm32f4-discovery device). I've followed the documentation (targets/os/zephyr/README.md) and everything worked fine - my OS is Ubuntu 20.04. After flashing the binary to the board, the header of the demo app appeared as expected, while the prompt does not.

```
*** Booting Zephyr OS build zephyr-v2.7.0 ***

JerryScript build: Jan  3 2023 16:17:15
JerryScript API 3.0.0
Zephyr version 2.7.0
```

The prompt problem is solved by applying #5033 modification.