jerryscript-project/jerryscript #5036: Some errors at compile time

emith487, 2023-01-16:
1、Using ubantu20.04 server environment to run code, encountered some errors when compiling, may I ask what method can solve?

emith487, 2023-01-16:
- Scanning dependencies of target jerryscript.elf

make[2] : *** No rule to make target

make[2] : *** Waiting for unfinished jobs...

make[1] : ***  [CMakeFiles/Makefile2:2411: CMakeFiles/ierryscript.elf.dir/alll Error 2

make: *** [Makefile:130: all] Error 2

make failed with exit code 2

ossy-szeged, 2023-01-16:
Have you installed dependencies? tools/apt-get-install-deps.sh
And then python tools/build.py should work without error.

akosthekiss, 2023-01-16:
Please, file a proper bug report with all the details requested in the template filled in:

- JerryScript revision: Identify the git hash(es) or tag(s) where the issue was observed.
- Build platform: Name the build platform. E.g., copy the output of `echo "$(lsb_release -ds) ($(uname -mrs))"` (on Linux), `echo "$(sw_vers -productName) $(sw_vers -productVersion) ($(uname -mrs))"` (on macOS), or `python -c "import platform; print(platform.platform())"` (should work everywhere).
- Build steps: Describe how to build JerryScript. Give all the necessary details of the build (e.g., environment variables, command(s), profile, command line options, etc.). E.g.: `tools/build.py --clean --debug` or `mkdir build && cmake -H. -Bbuild && make -C build`
- Build log

(I've edited the bug template for brevity. Please, find the complete bug report template at: https://github.com/jerryscript-project/jerryscript/blob/master/.github/ISSUE_TEMPLATE/bug_report.md . It is usually already pre-filled in the text editor box when you start reporting a bug. Just don't delete it next time.)

emith487, 2023-01-17:
> 
Have relied on，buddy

emith487, 2023-01-17:
The following commands are used.

-- sh jerryscript/tools/apt-get-install-deps.sh
-- python -m pip install --user -r esp-idf/requirements.txt
-- tar -xvf xtensa-esp32-elf-gcc8_4_0-esp-2021r2-patch5-linux-amd64.tar.gz 
-- python esp-idf/tools/idf.py --project-dir jerryscript/targets/baremetal-sdk/espressif --build-dir jerryscript/build/esp-idf set-target esp32 all

ossy-szeged, 2023-01-17:
ESP IDF build is broken for a while, similar issue - #5034

emith487, 2023-01-17:
> Have you installed dependencies? tools/apt-get-install-deps.sh And then python tools/build.py should work without error.

Buddy，I forgot to tell you,Last time you asked,path5 should be used.

emith487, 2023-01-17:
> ESP IDF build is broken for a while, similar issue - #5034

I know your problem.

emith487, 2023-01-17:
> ESP IDF build is broken for a while, similar issue - #5034

I checked with the original factory,I have asked the original factory about this link, which is caused by the version mismatch of tool chain under idf version,He suggested I replace the tool chain.https://github.com/espressif/crosstool-NG/tags

akosthekiss, 2023-01-17:
Buddy, I know your problem. I just forgot to tell you: last time you asked, we were not buddies.

closed 2023-01-17 by akosthekiss