jerryscript-project/jerryscript #5042: String.startsWith fails with -Ofast

ReimuNotMoe, 2023-02-11:
When JerryScript is compiled with `-Ofast`, `String.startsWith`  won't work correctly anymore.

This problem doesn't happen with lower optimization levels.

**I noticed some functions in `ecma-builtin-helpers.c` are using `ecma_number_t` in for loops. Is this really necessary? Since `ecma_number_t` is defined as `double`, this can [cause some problems](https://stackoverflow.com/questions/16595668/any-risk-of-using-float-variables-as-loop-counters-and-their-fractional-incremen). I'm not sure if it's the case but I suspect it's related.**

#### JerryScript revision
1a2c04763aba49f52b1537acd3730098c873511c

#### Build platform
```
Linux (none) 5.19.14-x1000 #477 PREEMPT Wed Dec 7 04:29:41 CST 2022 mips GNU/Linux
gcc version 11.3.0 (Buildroot 2022.02.2)
```

#### Build steps

```
set(CFLAGS_COMMON -march=mips32r2 -mno-interlink-compressed -mno-shared -Ofast --param=l1-cache-size=16 --param=l1-cache-line-size=32 --param=l2-cache-size=128)
add_compile_options(${CFLAGS_COMMON})
add_link_options(${CFLAGS_COMMON} -Wl,--gc-sections)

set(JERRY_ERROR_MESSAGES ON CACHE BOOL "" FORCE)
set(JERRY_LINE_INFO ON CACHE BOOL "" FORCE)
set(JERRY_MEM_STATS ON CACHE BOOL "" FORCE)
set(JERRY_CMDLINE OFF CACHE BOOL "" FORCE)
set(JERRY_EXTERNAL_CONTEXT OFF CACHE BOOL "" FORCE)
```

#### Test case
```js
s = "LV_PART_MAIN";
s.startsWith("LV_PART_");
```

#### Output
`false`

#### Expected behavior
`true`

nekopsykose, 2023-02-11:
> -Ofast

you built using options documented to break the standard, and you're surprised something broke?

ReimuNotMoe, 2023-02-11:
> you built using options documented to break the standard, and you're surprised something broke?

I'm not the only one with fault here since many [incorrect coding practises](http://codepad.org/c5xVNVDA) can be discovered with high optimization levels. So it's a problem worth discussing about.

zherczeg, 2023-02-12:
Isn't mips officially abandoned? Maybe unmaintained compiler tools cause issues.

nekopsykose, 2023-02-12:
> So it's a problem worth discussing about.  

sure- but only insofar as something being wrong because it is wrong based on some level of correctness, not related to Ofast itself.

> Isn't mips officially abandoned? Maybe unmaintained compiler tools cause issues.  

maybe, maybe not, but for a simple test, i couldn't reproduce this on x86_64-linux-musl with either gcc12.2 or clang15 with -Ofast.

(that, of course, doesn't rule out a correctness issue- UB code does not always fail all the time on every platform, but it's possible everything is correct and this is purely a bad compiler/platform issue, as noted above)

ReimuNotMoe, 2023-02-12:
@zherczeg 

> Isn't mips officially abandoned? Maybe unmaintained compiler tools cause issues.

No it's not. Waves sold all MIPS IPs to CIP United and the development is being continued. Nowadays the latest 5G basebands in Mediatek chipsets still uses MIPS architecture.

The gcc version is 11.3.0, which should be up-to-date enough.

What's your opinion about using `double` in for loops? Do you think they're necessary or should be replaced to integers?

ReimuNotMoe, 2023-02-12:
> @zherczeg
> 
> > Isn't mips officially abandoned? Maybe unmaintained compiler tools cause issues.
> 
> No it's not. Waves sold all MIPS IPs to CIP United and the development is being continued. Nowadays the latest 5G basebands in Mediatek chipsets still uses MIPS architecture.
> 
> The gcc version is 11.3.0, which should be up-to-date enough.
> 
> What's your opinion about using `double` in for loops? Do you think they're necessary or should be replaced with integers?

zherczeg, 2023-02-12:
Do you have a link for that? It probably depends on the case.

ReimuNotMoe, 2023-02-12:
> Do you have a link for that? It probably depends on the case.

[ecma-builtin-helpers.c#L492](https://github.com/jerryscript-project/jerryscript/blob/master/jerry-core/ecma/builtin-objects/ecma-builtin-helpers.c#L492)

zherczeg, 2023-02-12:
This one can be changed. Though it should not cause any issues.

ClassicOldSong, 2023-02-12:
`indexOf` has the same problem, most likely they use the same mechanism for indexing