jerryscript-project/jerryscript #5051: segmentation fault in jerryscript

zhunki, 2023-03-14:
the following js cause a segmentation fault in jerryscript.

```
function test(){
	function f(){
		var f = test();
	   	h = f(); 
	}
	return f;
}
let a = test();
c = a();
```

marckwei, 2023-04-06:
###### JerryScript revision
1a2c04763aba49f52b1537acd3730098c873511c

###### Build platform
Name the build platform. E.g., copy the output of
Ubuntu Lunar Lobster (development branch) (Linux 5.15.0-67-generic x86_64)

###### Build steps
```sh
export CC=clang
tools/build.py  --debug
```

###### Test case
function test(){
	function f(){
		var f = test();
	   	h = f(); 
	}
	return f;
}
let a = test();
c = a();

###### Execution steps
./build/bin/jerry  test.js

###### Output
segmentfault
###### Backtrace
==51718==ERROR: AddressSanitizer: stack-overflow on address 0x7ffd01b29ff8 (pc 0x5596f1c550b6 bp 0x7ffd01b2a3e0 sp 0x7ffd01b29fe0 T0)
    #0 0x5596f1c550b6 in vm_loop /home/zxw/jerryscript/jerry-core/vm/vm.c:970
    #1 0x5596f1c676fd in vm_execute /home/zxw/jerryscript/jerry-core/vm/vm.c:5211
    #2 0x5596f1c55050 in vm_run /home/zxw/jerryscript/jerry-core/vm/vm.c:5312
    #3 0x5596f1c1aafc in ecma_op_function_call_simple /home/zxw/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1176
    #4 0x5596f1c1d62d in ecma_op_function_call /home/zxw/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1406
    #5 0x5596f1c678b6 in ecma_op_function_validated_call /home/zxw/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1371
    #6 0x5596f1c678b6 in opfunc_call /home/zxw/jerryscript/jerry-core/vm/vm.c:758
    #7 0x5596f1c678b6 in vm_execute /home/zxw/jerryscript/jerry-core/vm/vm.c:5217
    #8 0x5596f1c55050 in vm_run /home/zxw/jerryscript/jerry-core/vm/vm.c:5312
    #9 0x5596f1c1aafc in ecma_op_function_call_simple /home/zxw/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1176
    #10 0x5596f1c1d62d in ecma_op_function_call /home/zxw/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1406
    #11 0x5596f1c678b6 in ecma_op_function_validated_call /home/zxw/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1371
    #12 0x5596f1c678b6 in opfunc_call /home/zxw/jerryscript/jerry-core/vm/vm.c:758
...
...
...
    #366 0x5596f1c678b6 in opfunc_call /home/zxw/jerryscript/jerry-core/vm/vm.c:758
    #367 0x5596f1c678b6 in vm_execute /home/zxw/jerryscript/jerry-core/vm/vm.c:5217
    #368 0x5596f1c55050 in vm_run /home/zxw/jerryscript/jerry-core/vm/vm.c:5312
    #369 0x5596f1c1aafc in ecma_op_function_call_simple /home/zxw/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1176
    #370 0x5596f1c1d62d in ecma_op_function_call /home/zxw/jerryscript/jerry-core/ecma/operations/ecma-function-object.c:1406

SUMMARY: AddressSanitizer: stack-overflow /home/zxw/jerryscript/jerry-core/vm/vm.c:970 in vm_loop
==51718==ABORTING

carnil, 2023-04-26:
CVE-2023-30414 was assigned for this issue.

ossy-szeged, 2024-11-21:
Dup of #5039