jerryscript-project/jerryscript #5052: jerry crashed while running the following code.

marckwei, 2023-03-15:
###### JerryScript revision
1a2c04763aba49f52b1537acd3730098c873511c

###### Build platform
Name the build platform. E.g., copy the output of
Ubuntu Lunar Lobster (development branch) (Linux 5.15.0-67-generic x86_64)

###### Build steps
Describe how to build JerryScript. Give all the necessary details of the build
(e.g., environment variables, command(s), profile, command line options, etc.).

E.g.:
```sh
export CC=clang
tools/build.py  --debug
```

###### Test case
function f0(a1, a2, ...a3) {
    class C4 extends a1 {
    }
    var v5 = new C4(C4);
    return C4;
}
f0(f0);

###### Execution steps
./build/bin/jerry  test.js

###### Output
segmentfault

carnil, 2023-04-26:
CVE-2023-30410 appears to have been assigned for this issue.

ossy-szeged, 2024-11-21:
dup of #5039